project-chip/connectedhomeip #20718: [SVE#1][TC-BOOL-2.2] chiptool report "Segmentation fault" when subscribing event

tanyue518, 2022-07-14:
DUT: Contact Sensor
Cluster: Boolean State (Server)
Case: TC-BOOL-2.2,  Primary functionality with server as DUT, Step 4a

this the reproduction procedures using the chiptool under TH image apps folder
1) commission the DUT to network
2) sudo ./chip-tool interactive start
3) booleanstate subscribe-event-by-id 0x0 1 100 1 1

chiptool report "Segmentation fault" after subscribing event command

sstruchtrup, 2022-07-14:
I am seeing the same problem. Seems to affect all clusters/attributes.

tanyue518, 2022-07-14:
yes, I can fix the issue by rebuilding, but ATL should another TH image from CSA

bzbarsky-apple, 2022-07-14:
@cjandhyala What SHA is being used for the chip-tool image here?

andy31415, 2022-07-14:
@tanyue518 please provide more info:

- DUT Contact sensor ... on what platform? Does this occur on any platform so we can check (i.e. can i reproduce this on linux with all-clusters app or similar?)
- what SHA is this using?

andy31415, 2022-07-14:
If this is fixed by "update to latest code" this is not a cert blocker as it is already fixed. Am I understanding the status right?

bzbarsky-apple, 2022-07-14:
The SHA involved is likely caf62f709e0a0fdf41b0d3a1652cc26bc46a960a

@andy31415 @cjandhyala tells me that they have been seeing random crashes on all sort of revisions here, including "latest code".  So there may still be an issue here.

bzbarsky-apple, 2022-07-14:
That said, I just compiled chip-tool from that SHA (on Mac, with ASAN) and I am not seeing a crash when using all-clusters-app.  This might depend on the exact data the DUT is sending.

@tanyue518 can you please attach (not paste) a complete logs for both sides here?  Especially the chip-tool log showing the incoming payloads (if any) and whatever else happens before the crash.

tanyue518, 2022-07-15:
sorry for late reply, the Contact Sensor is on NXP K32W061 platform, I don't know the SHA for chiptool, but the chiptool is the default one under apps folder of sve1.0 TH image.

vivien-apple, 2022-07-15:
I wish we have logs for both sides. There are some interesting logs there. For example it seems that some unexpected packets are received.
```
1657878171.170570][2451:2456] CHIP:EM: Flushed pending ack for MessageCounter:132062254 on exchange 54186i
[1657878171.516869][2451:2456] CHIP:EM: Received message of type 0x10 with protocolId (0, 0) and MessageCounter:132062255 on exchange 54186i
[1657878171.516938][2451:2456] CHIP:EM: OnMessageReceived failed, err = ../../src/messaging/ExchangeMgr.cpp:274: CHIP Error 0x00000070: Unsolicited msg with originator bit clear
[1657878171.533943][2451:2456] CHIP:EM: Received message of type 0x10 with protocolId (0, 0) and MessageCounter:132062256 on exchange 54186i
[1657878171.534050][2451:2456] CHIP:EM: OnMessageReceived failed, err = ../../src/messaging/ExchangeMgr.cpp:274: CHIP Error 0x00000070: Unsolicited msg with originator bit clear
[1657878171.566911][2451:2456] CHIP:IN: Received a duplicate message with MessageCounter:132062254 on exchange 54186i
[1657878171.566991][2451:2456] CHIP:EM: Received message of type 0x40 with protocolId (0, 0) and MessageCounter:132062254 on exchange 54186i
[1657878171.567075][2451:2456] CHIP:EM: Generating StandaloneAck via exchange: 54186i
[1657878171.567137][2451:2456] CHIP:EM: Forcing tx of solitary ack for duplicate MessageCounter:132062254 on exchange 54186i
[1657878171.567195][2451:2456] CHIP:EM: Sending Standalone Ack for MessageCounter:132062254 on exchange 54186i
```

bzbarsky-apple, 2022-07-19:
Hmm.  @tanyue518 If you run under a debugger can you at least see where chip-tool is crashing, exactly?

tanyue518, 2022-07-24:
this issue is fixed for TH 1.3 image

closed 2022-07-24 by tanyue518