project-chip/connectedhomeip #20719: Test_TC_DGTHREAD_2_1 the PICS values for the attributes are off by one. [SVE1]

markus-becker-tridonic-com, 2022-07-14:
#### Problem

In Test_TC_DGTHREAD_2_1 the PICS values for the attributes are off by one.

E.g. the channel attribute is 0, but the PICS value is DGTHREAD.S.A0001, https://github.com/project-chip/connectedhomeip/blob/master/src/app/tests/suites/certification/Test_TC_DGTHREAD_2_1.yaml#L31-L37.

 #### Proposed Solution

Reduce the numbers in the PICS by one.

kowsisoundhar12, 2022-07-18:
@markus-becker-tridonic-com updated DGTHREAD scripts PICS values in this [PR](https://github.com/project-chip/connectedhomeip/pull/20876),please verify it once and close this issue

markus-becker-tridonic-com, 2022-07-19:
Verified. Thanks.

Note that not all attributes are not covered anymore: https://github.com/CHIP-Specifications/chip-test-plans/issues/1818.

closed 2022-07-19 by markus-becker-tridonic-com