project-chip/connectedhomeip #20731: TC-BRBINFO-1.1 - Feature-map attribute should be implemented

sowmyassp, 2022-07-14:
**Issue :** 
While Executing the behavior of the global mandatory feature-map attribute  of the Bridged Basic Information cluster, throws error code as 0x86(unsupported attribute).

**Commit used** : 0390a8d33b1d1c709fcc4499eab6e132428ad5e2

- Chip-tool - RPI-4, 8GB RAM
- DUT - RPI - RPI-4, 8GB RAM
- Platform : linux platform
- Dut application: bridge-app
- Expected behavior : Pass with status(0x0)

**Actual behavio**r : Fail with status 0x86(unsupported attribute)
[image]

**Steps to reproduce :**
./chip-tool bridgeddevicebasic read feature-map 1 3

**Test-plan reference :**
https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/cluster/bridged_device_basic_information.adoc#tc-brbinfo-1-1-global-attributes-for-bridged-device-basic-information-cluster-cluster-dut-server

andy31415, 2022-07-14:
I think we need to explain/justify cert blocker status a bit here.

Cert blocker in my mind means "Certification cannot prove that the SDK can be used to build an application that can be certified". In this case, we have feature map attributes working in other apps, so the SDK can be used to build certifiable applications.

Bridge app in particular seems not to be updated fully, so this is an issue and we may chose to do so before 1.0, however in terms of certification blocking, I do not believe this is so.

andy31415, 2022-07-14:
In generally I am trying to avoid code SDK developers being on the hook to develop fully certifiable applications for all possible applications (we have a lot of them already: light, lock, all-clusters, all-clusters-minimal, several tv bits, bridge, ota provider and requestor, shell, thermostat, temperature measurement, window, pump .... etc).

cjandhyala, 2022-07-15:
@andy31415 Cert blocker is also for something that is failing a certification test case that need fix in SDK and example app, so that CSG can verify the test ones. In this case its blocking a cert TC TC-BRBINFO-1.1.

woody-apple, 2022-07-19:
Cert Blocker Review: Given bridge app is used for verification, we need to make sure the feature map is correct.

franck-apple, 2022-07-21:
@jwinder-ca is looking at this.

PR #21061 (2022-07-21): Added support for the bridged device basic feature-map attribute.
#### Problem
Fixes #20731 Feature-map attribute should be implemented

#### Change overview
Added support for the Bridged Device Basic feature-map attribute to the Linux bridge-app.

#### Testing
Manually tested using chip-tool. 
1. Start bridge app and wait for initialization to complete
2. Run chip-tool pairing
    chip-tool pairing ethernet 12344321 20202021 3840 localhost 5540
3. Read the Bridged Device Basic feature-map attribute using chip-tool
    chip-tool bridgeddevicebasic read feature-map 12344321 3
PR #21083 (2022-07-21): Added support for the bridged device basic feature-map attribute.
#### Problem
Fixes #20731 Feature-map attribute should be implemented

#### Change overview
Added support for the Bridged Device Basic feature-map attribute to the Linux bridge-app.

#### Testing
Manually tested using chip-tool. 
1. Start bridge app and wait for initialization to complete
2. Run chip-tool pairing
    chip-tool pairing ethernet 12344321 20202021 3840 localhost 5540
3. Read the Bridged Device Basic feature-map attribute using chip-tool
    chip-tool bridgeddevicebasic read feature-map 12344321 3
closed 2022-07-21 by bzbarsky-apple