project-chip/connectedhomeip #20734: [SVE][TC-CNET-4.1] thread device’s PICS should NOT test WiFi test case

yeccc, 2022-07-14:
#### Problem
<what's wrong or missing, please include any applicable:
 * expected behavior
 * actual behavior
 * steps to reproduce
 * system configuration
 * screenshots, images, logs, etc.
 >
[image]

 #### Proposed Solution
<suggested fix, suggested enhancement>

bzbarsky-apple, 2022-07-14:
@cjandhyala

franck-apple, 2022-07-18:
removing the 'request cert blocker' label, as it's a script issue.

woody-apple, 2022-07-19:
Cert Blocker Review: This is not an SDK issue. Can you please make sure this is filed in the test harness repo.

closed 2022-07-19 by woody-apple

franck-apple, 2022-07-19:
test harness repo: https://github.com/CHIP-Specifications/chip-certification-tool