project-chip/connectedhomeip #20769: Sensibly convert CHIP_ERROR to IM Status when returning a `StatusIB`

mrjerryjohns, 2022-07-14:
#### Problem

When doing IM operations like reads, cluster impls today still continue to send back CHIP_ERROR with a non-IM status code embedded in it. This eventually gets coarsely converted to an IM `Failure` status code, the least descriptive of all IM status codes.

While one option is to fix all cluster impls pervasively, this can be laborious.

Another option is to add a converter from `CHIP_ERROR` to IM error that converts most standard CHIP errors in `StatusIB::InitFromChipError`. This should also accept an arg to indicate the IM operation to sensibly map some errors differently depending on whether it's an invoke/read/write, etc.

bzbarsky-apple, 2022-07-15:
The reason I have not done this so far is that I feel like it would give people a false sense of security that they are returning the right things, when they are propagating error codes from some underlying implementation and those can change on them at any time without them noticing....

Maybe that's OK.  But it worries me a little.

bzbarsky-apple, 2022-07-15:
For example, in #20736 CHIP_ERROR_NOT_FOUND is being mapped to ConstraintError but other places would want to map it differently, depending on what CHIP_ERROR_NOT_FOUND means in the specific APIs they're calling.