project-chip/connectedhomeip #20787: Single read-event causes several error responses instead of one

bluebin14, 2022-07-15:
#### Problem
One single read-event causes as many UNSUPPORTED_ACCESS status responses as there are events in history, thereby giving out the exact number of events to unauthorized peer. This behavior does not match the spec, which has a single UNSUPPORTED_ACCESS for the one concrete path in a case like this. Observed with TC-ACL-2.9 but it is a general problem that can happen with any cluster.

`//TC-ACL-2.9
// STEP 10
./chip-tool accesscontrol read-event access-control-entry-changed 12344321 0
...
[1657826747.396451][18408:18413] CHIP:TOO: Response Failure: IM Error 0x0000057E: General error: 0x7e (UNSUPPORTED_ACCESS)
[1657826747.396517][18408:18413] CHIP:TOO: Response Failure: IM Error 0x0000057E: General error: 0x7e (UNSUPPORTED_ACCESS)
[1657826747.396581][18408:18413] CHIP:TOO: Response Failure: IM Error 0x0000057E: General error: 0x7e (UNSUPPORTED_ACCESS)
[1657826747.396645][18408:18413] CHIP:TOO: Response Failure: IM Error 0x0000057E: General error: 0x7e (UNSUPPORTED_ACCESS)
[1657826747.396707][18408:18413] CHIP:TOO: Response Failure: IM Error 0x0000057E: General error: 0x7e (UNSUPPORTED_ACCESS)
[1657826747.396770][18408:18413] CHIP:TOO: Response Failure: IM Error 0x0000057E: General error: 0x7e (UNSUPPORTED_ACCESS)
[1657826747.396834][18408:18413] CHIP:TOO: Response Failure: IM Error 0x0000057E: General error: 0x7e (UNSUPPORTED_ACCESS)
[1657826747.396898][18408:18413] CHIP:TOO: Response Failure: IM Error 0x0000057E: General error: 0x7e (UNSUPPORTED_ACCESS)
[1657826747.396962][18408:18413] CHIP:TOO: Response Failure: IM Error 0x0000057E: General error: 0x7e (UNSUPPORTED_ACCESS)
[1657826747.397025][18408:18413] CHIP:TOO: Response Failure: IM Error 0x0000057E: General error: 0x7e (UNSUPPORTED_ACCESS)
[1657826747.397088][18408:18413] CHIP:TOO: Response Failure: IM Error 0x0000057E: General error: 0x7e (UNSUPPORTED_ACCESS)
[1657826747.397149][18408:18413] CHIP:TOO: Response Failure: IM Error 0x0000057E: General error: 0x7e (UNSUPPORTED_ACCESS)`

 #### Proposed Solution
Only one error response should be returned.

bzbarsky-apple, 2022-07-15:
It sounds like we are doing ACL checks on a per-event basis, not per-path basis.  The testcase here is basically:

1. Set ACLs so you have no access.
2. Try to read a single concrete event path.

Looking at our code, it's just structured _very_ differently from the spec.  The spec loops over paths, puts in errors for the concrete ones that are not allowed and drops them, then starts looping over events and matching them to remaining paths.  We just loop over events and match them to paths, so we keep matching against the same disallowed path over and over.

bluebin14, 2022-07-16:
As an additional side effect of the above, if ACL extension is not set then read-event access-control-extension-changed succeeds even without access, which fails certification of TC-ACL-2.9 step 11.

yunhanw-google, 2022-07-18:
@erjiaqing , could you take care of this ticket? thanks

erjiaqing, 2022-07-21:
After checking the code, the solution should be:

Before actually generating events, iterate through the interested paths of the read handler, and if we don't have enough privilege for reading the event, emit UNSUPPORTED_ACCESS and drop the path.

One thine we need to take care of is we may run out of buffer when encoding these status IBs.

bluebin14, 2022-07-21:
@erjiaqing emitting UNSUPPORTED_ACCESS per path as opposed to per event actually needs much less buffer so it should work.

cjandhyala, 2022-07-29:
Aswathy to verify and close the ticket

woody-apple, 2022-07-29:
Cert Blocker Review: Closing this, given we believe the merged PR resolves this issue.

closed 2022-07-29 by woody-apple