project-chip/connectedhomeip #20789: [TC-ACL-2.4]: Test case Step 43: Expected output is constraint error , But the actual result is Failure

aswathygrl, 2022-07-15:
Test Case ID
TC-ACL-2.4
Describe Issue:
Step 43 :expected outcome is constraint error but result is Failure.
Command to Verify:./chip-tool accesscontrol write acl '[{"fabricIndex": 1, "privilege": 5, "authMode": 2, "subjects": [112233], "targets": null}, {"fabricIndex": 1, "privilege": 3, "authMode": 2, "subjects": null, "targets":[{ "cluster": 11, "endpoint": 22, "deviceType": 33 }]}]' 1 0
Specification References:
https://github.com/CHIP-Specifications/connectedhomeip-spec/blob/master/src/data_model/ACL-Cluster.adoc#534-targets
Test Plan Reference:-
https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/cluster/AccessControl.adoc#tc-acl-2-4-acl-attribute

bzbarsky-apple, 2022-07-15:
@aswathygrl Please always include the SHA the problem occurred on....

I believe this will be fixed by #20736

woody-apple, 2022-07-19:
Cert Blocker Review: Assigning to @cjandhyala as well for retest.

cjandhyala, 2022-07-22:
per @aswathygrl this is working. hence closing the issue.

closed 2022-07-22 by cjandhyala