project-chip/connectedhomeip #20790: [TC-ACL-2.4]: Test case ,Step 40: Expected output is constraint error , But the actual result is Success.

aswathygrl, 2022-07-15:
Test Case ID
TC-ACL-2.4
Describe Issue:
Step 40 :expected outcome is constraint error but result is Success.
Command to Verify:./chip-tool accesscontrol write acl '[{"fabricIndex": 1, "privilege": 5, "authMode": 2, "subjects": [112233], "targets": null}, {"fabricIndex": 1, "privilege": 3, "authMode": 2, "subjects": null, "targets":[{ "cluster": null, "endpoint": 255, "deviceType": null }]}]' 1 0
Specification References:
https://github.com/CHIP-Specifications/connectedhomeip-spec/blob/master/src/data_model/ACL-Cluster.adoc#534-targets
Test Plan Reference:-
https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/cluster/AccessControl.adoc#tc-acl-2-4-acl-attribute

bzbarsky-apple, 2022-07-15:
This testcase is wrong, as far as I can tell.  Why should this fail?  @cjandhyala

sankarselvam, 2022-07-18:
@bzbarsky-apple and @andy31415  As per the spec, Target Struct shall not contain both an Endpoint field and a DeviceType field.  Table 80 and Table 81 in the spec section 9.10.5., (In PDF). Github section of spec is referred above.  
Also the error handling section 9.10.6 mentions, Invalid Target element & both Endpoint and DeviceType fields present - will result in Constraint Error. 
Please correct us, if our understanding is wrong. 

In case, if we say, the cluster and devicetype is null in the above command, hence the response is  success.

For the below commands, DUT should respond with Constraint error per spec from my understanding. 

 ./chip-tool accesscontrol write acl '[{"fabricIndex": 1, "privilege": 5, "authMode": 2, "subjects": [112233], "targets": null}, {"fabricIndex": 1, "privilege": 3, "authMode": 2, "subjects": null, "targets":[{ "cluster": 5, "endpoint": 255, "deviceType": 2}]}]' 1 0

 ./chip-tool accesscontrol write acl '[{"fabricIndex": 1, "privilege": 5, "authMode": 2, "subjects": [112233], "targets": null}, {"fabricIndex": 1, "privilege": 3, "authMode": 2, "subjects": null, "targets":[{ "cluster": null , "endpoint": 255, "deviceType": 2}]}]' 1 0

[image]

andy31415, 2022-07-18:
So test case should be corrected right?

Because the initial summary said the test to expect failure is:

```
./chip-tool accesscontrol write acl \
    '[{"fabricIndex": 1, "privilege": 5, "authMode": 2, "subjects": [112233], "targets": null}, \
    {"fabricIndex": 1, "privilege": 3, "authMode": 2, "subjects": null, "targets":[\
        { "cluster": null, "endpoint": 255, "deviceType": null }]}]' \
    1 0
```

Which seems like endpoint is set but device type is not.

bzbarsky-apple, 2022-07-19:
>  Target Struct shall not contain both an Endpoint field and a DeviceType field. 

What the spec is _trying_ to say is that those fields can't both be non-null at the same time.

I agree that it could be clearer about that.  Please raise a spec issue as needed.

sankarselvam, 2022-07-19:
Thanks @bzbarsky-apple  Aswathy raised a Spec issue for the devicetype and endpoint. 
@aswathygrl  The test case needs to be modified for its expected value. 

Step 40 :expected outcome is constraint error but result is Success.
Command to Verify:./chip-tool accesscontrol write acl '[{"fabricIndex": 1, "privilege": 5, "authMode": 2, "subjects": [112233], "targets": null}, {"fabricIndex": 1, "privilege": 3, "authMode": 2, "subjects": null, "targets":[{ "cluster": null, "endpoint": 255, "deviceType": null }]}]' 1 0

When I deep dive,  for the test step 40, for the given command, test case should get success. Please raise a test plan issue and update the expected outcome.

aswathygrl, 2022-07-20:
@sankarselvam I will raise a test plan issue on TC-ACL-2.4 ,Step 40.

closed 2022-07-20 by aswathygrl