project-chip/connectedhomeip #20791: [TC-CGEN-2.2]--AddTrustedRootCertificate command  Fails to add TrustedRootCertificate

Rajashreekalmane, 2022-07-15:
**#### Problem**
add-trusted-root-certificate command is not working, TH sends add-trusted-root-certificate to DUT then if TH read the trsusted root certificate attribute where it should list 2 entries but instead lists 1 entry
 
*** expected behavior:-**The number of items in the returned list is numTrustedRootsOriginal + 1
 *** actual behavior:-**
[image]

  *** steps to reproduce:--**
To generate TrustedRootCertificate, need to buld chip-cert in connectedhomeip by giving below mentioned commands
1. gn gen --check out/debug
2. ninja -v -C out/debug chip-cert
Once build completes 
3.  ./chip-cert gen-cert --type r --subject-chip-id CACACACA00000001 --valid-from "2020-10-15 14:23:43" --lifetime 7305 --out-key Chip-Root-Key.pem --out Chip-Root-Cert.pem --out-format x509-pem
4. cat Chip-Root-Key.pem 
FTABAQAkAgE3AyQUABgmBIAigScmBYAlTTo3BiQUABgkBwEkCAEwCUEEcHhi5skO0neoie7A8cAE0AhLfjo1GZJLURJBw/MuMi24aVlrHrtZuwqe3E/wtVUK5kv5jWWTustZ9Jse8ZHyfDcKNQEpARgkAmAwBBRVQgQw34aJK7lbbOiFhZPiQdyq7DAFFFVCBDDfhokruVts6IWFk+JB3KrsGDALQLgPckNnCsuVE/LGmZ5GGWcBnc5+ZNLl91xUR6JFBYBOw6SEyNGUdWFkl1530ov3XhD+lbLcZ3IMgds0rCgaf0QY

-    ./chip-tool generalcommissioning  arm-fail-safe   60 1 1 0

-    ./chip-tool  operationalcredentials add-trusted-root-certificate FTABAQAkAgE3AyQUABgmBIAigScmBYAlTTo3BiQUABgkBwEkCAEwCUEEcHhi5skO0neoie7A8cAE0AhLfjo1GZJLURJBw/MuMi24aVlrHrtZuwqe3E/wtVUK5kv5jWWTustZ9Jse8ZHyfDcKNQEpARgkAmAwBBRVQgQw34aJK7lbbOiFhZPiQdyq7DAFFFVCBDDfhokruVts6IWFk+JB3KrsGDALQLgPckNnCsuVE/LGmZ5GGWcBnc5+ZNLl91xUR6JFBYBOw6SEyNGUdWFkl1530ov3XhD+lbLcZ3IMgds0rCgaf0QY  1  0

-   ./chip-tool  operationalcredentials read  trusted-root-certificates 1 0 

**system configuration**
Chip-tool - RPI-4, 8GB RAM
DUT - RPI - RPI-4, 8GB RAM
Platform : IP and wifi
Commit-ID: 0390a8d33b1d1c709fcc4499eab6e132428ad5e2
Dut application: all-clusters-app

**Test-Plan Reference:-**
https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/cluster/General_Commissioning.adoc#tc-cgen-2-2-armfailsafe-command-verification-dut-server

PFA log below:-
[TC-CGEM-2.2.txt](https://github.com/project-chip/connectedhomeip/files/9120085/TC-CGEM-2.2.txt)

bzbarsky-apple, 2022-07-15:
Yep, this is still buggy.  @tcarmelveilleux

bzbarsky-apple, 2022-07-15:
`OperationalCredentialsAttrAccess::ReadRootCertificates` needs to include the pending root from the fabric table if there is one and the AddNOC has not happened yet (so there is no FabricInfo for it yet).

woody-apple, 2022-07-19:
Cert Blocker Review: Assigning @bzbarsky-apple to take a look.

bzbarsky-apple, 2022-07-19:
> ./chip-tool operationalcredentials add-trusted-root-certificate [...]

@Rajashreekalmane that's not a valid command....

bzbarsky-apple, 2022-07-19:
And when I try `--out-format chip-hex`, the result fails server-side validation:
```
chip-tool operationalcredentials add-trusted-root-certificate hex:043784DDE69F1BCAB0AB1248CF0B074E9E12C81DB24A396339E76CE058C5348D6FEB053A126750CF914AC7E903DCBC078C92A002F194908D8CDF775C86119C3A130850ED77100DD1AC242EF6C33C715B65E375D5AC571819E60D76EEF7E060B23A 1 0
....
[1658267423499] [28928:32299064] CHIP: [ZCL] OpCreds: Failed AddTrustedRootCertificate request with IM error 0x85 (err = ../../examples/lock-app/linux/third_party/connectedhomeip/src/lib/core/CHIPTLVReader.cpp:543: CHIP Error 0x00000026: Wrong TLV type)
```

@turon @Rajashreekalmane what are the actual steps that should be happening here?

bzbarsky-apple, 2022-07-19:
OK, so if I do:
```
chip-cert gen-cert --type r  --out-key Chip-Root-Key.txt --out Chip-Root-Cert.txt --out-format chip-hex
```
and then get the cert from that file and do:
```
chip-tool operationalcredentials add-trusted-root-certificate hex:153001082B17D22370EF05182402013703271401000000CACACACA182604EF171B2726056EB5B94C3706271401000000CACACACA18240701240801300941043784DDE69F1BCAB0AB1248CF0B074E9E12C81DB24A396339E76CE058C5348D6FEB053A126750CF914AC7E903DCBC078C92A002F194908D8CDF775C86119C3A13370A35012901182402603004141A0D632703B7AC87D3A9C07CA57ED53638F881743005141A0D632703B7AC87D3A9C07CA57ED53638F8817418300B4004C13EC2AE2AD3609DBBBF8BF0D1F74E5B1C5A13B3543F76CF33E8F01A212BDA2EF659B75659EC7941507648960F4C39EDDFC11D86838E15C9BA55A68CEEF1BE18 1 0
```
then I can reproduce.
```

closed 2022-07-20 by woody-apple

doublemis1, 2022-08-09:
@woody-apple 
Is there any plan to apply the above fix to SVE branch?