project-chip/connectedhomeip #20794: [TC-ACL-2.4]: Test case ,Step 32: Expected output is constraint error , But the actual result is Failure

aswathygrl, 2022-07-15:
Test Case ID
TC-ACL-2.4
Describe Issue:
Step 32 :expected outcome is constraint error but result is Failure.
Command to Verify: ./chip-tool accesscontrol write acl '[{"fabricIndex": 1, "privilege": 5, "authMode": 2, "subjects": [112233], "targets": null}, {"fabricIndex": 1, "privilege": 6, "authMode": 2, "subjects": null, "targets":null}]' 1 0
Specification References:
https://github.com/CHIP-Specifications/connectedhomeip-spec/blob/master/src/data_model/ACL-Cluster.adoc#error-handling
Test Plan Reference:-
https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/cluster/AccessControl.adoc#tc-acl-2-4-acl-attribute

andy31415, 2022-07-15:
@bzbarsky-apple Testing it now.

I believe our privilege mappins are bit-fields, so 6 seemed to pass for me, but I have to try again. For this one I need to understand if the test case is bad or the code has bugs or my manual test was bad.

andy31415, 2022-07-15:
Re-tested and it is indeed fixed. I think I had an error in my manual test before. I will mark it as such.

@aswathygrl sorry for making you break these up... I really thought that my PR did not fix them, but it turns out I made manual errors when double-checking.

PR #20890 (2022-07-18): Fix ACL constraint error (return constraint error instead of failure when the ACL entry is invalid)
#### Problem
`FAILURE` reported instead of `CONSTRAINT_ERROR`

Fixes #20672
Fixes #20794 
Fixes #20796
Fixes #20797 (NOTE: uses 0xFFFF_FFFD_0000_0000 and NOT the original test value as that seemed odd)
Fixes #20798

#### Change overview
Fix return code
Add unit tests for automation

I could *NOT* add tests for the 64bit invalid subjects due to zap limitation (was later explained that I need to use quotes). In the mean time, the tests not covered were broken out in separate bugs that will be fixe [... cut]
closed 2022-07-18 by woody-apple