project-chip/connectedhomeip #20797: [TC-ACL-2.4]: Test case ,Step 36: Expected output is constraint error , But the actual result is success

aswathygrl, 2022-07-15:
Test Case ID
TC-ACL-2.4
Describe Issue:
Step 36 :expected outcome is constraint error but result is Success.
Command to Verify:  ./chip-tool accesscontrol write acl '[{"fabricIndex": 1, "privilege": 5, "authMode": 2, "subjects": [112233], "targets": null}, {"fabricIndex": 1, "privilege": 3, "authMode": 2, "subjects": [42949672930000000], "targets": null}]' 1 0
Test Plan Reference:-
https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/cluster/AccessControl.adoc#tc-acl-2-4-acl-attribute

andy31415, 2022-07-15:
@aswathygrl - what is 42949672930000000

Test case says `0xFFFFFFFD_BAD_NONE0000000` (looks like a typo? is it?) and 42949672930000000 in hex looks like 98967ffe363c80

andy31415, 2022-07-15:
@aswathygrl I think 42949672930000000 is not ok.

I think this should be either deleted or use 0xFFFF_FFFD_0000_0000 (i.e. 18446744060824649728)

andy31415, 2022-07-15:
In my fix PR, I marked that this is fixed, but assuming test value is 0xFFFF_FFFD_0000_0000

aswathygrl, 2022-07-18:
@andy31415  I  have provided this value (0xFFFF_FFFD_0000_0000 (i.e. 18446744060824649728) but still I am getting failure in the below commitid. 

> commitid   aa9457e6b94b735076dff6297176183bf9780177

PR #20890 (2022-07-18): Fix ACL constraint error (return constraint error instead of failure when the ACL entry is invalid)
#### Problem
`FAILURE` reported instead of `CONSTRAINT_ERROR`

Fixes #20672
Fixes #20794 
Fixes #20796
Fixes #20797 (NOTE: uses 0xFFFF_FFFD_0000_0000 and NOT the original test value as that seemed odd)
Fixes #20798

#### Change overview
Fix return code
Add unit tests for automation

I could *NOT* add tests for the 64bit invalid subjects due to zap limitation (was later explained that I need to use quotes). In the mean time, the tests not covered were broken out in separate bugs that will be fixe [... cut]