project-chip/connectedhomeip #20803: [SQA] [Thread]TC-CNET-4.12 Getting Run Command Failure after connect-network with WIFI_2ND_ACCESS_POINT

Survensa, 2022-07-15:
## Problem
Steps 2 - 7:  remove WIFI_1ST_ACCESS_POINT, add WIFI_2ND_ACCESS_POINT, Connect to WIFI_2ND_ACCESS_POINT
After Step 7 :  `./chip-tool networkcommissioning connect-network hex:1161111128222222 1 0 --Breadcrumb 2` not able to proceed further due to run command failure.

 ### Actual behavior after connect-network with WIFI_2ND_ACCESS_POINT

>  [1657887831.463607][3486:3486] CHIP:TOO: Run command failure: ../../commands/common/CHIPCommand.cpp:423: CHIP Error 0x00000032: Timeout

 ### Steps to reproduce

1.  Remove th_xpan network  {  `./chip-tool networkcommissioning  remove-network hex:1111111152222222 22 0` }
2.  Add THREAD_2ND_OPERATIONALDATASET { `./chip-tool networkcommissioning add-or-update-thread-network hex:0e080000000000010000000300000f35060004001fffe0020811111111282222220708fd5930b7d8b62f58051000112233445562778899aabbccddeeff030e4f70656e54687265746444656d6f010212340410005ed405199188e1a5c0dd607282d7480c0402a0fff8 22 0 (second network dataset value)` }
3. Read the network {  ./chip-tool networkcommissioning read networks 22 0 }

### System configuration

- > DUT : All-cluster-app ( RPI)
- > CNTRL : Chiptool ( RPI)
- > Network : Ble-Wifi

#### Log : [CNET Thread.txt](https://github.com/project-chip/connectedhomeip/files/9121573/CNET.Thread.txt)

bzbarsky-apple, 2022-07-15:
After the device connects to the second network, it can't communicate with the chip-tool, which is on the original network, right?  So the behavior here seems completely expected...  @Survensa

cecille, 2022-07-18:
Agree - the omission of the verification step here in the test plan is intentional. chip-tool won't receive a response on its network if the network has been changed.

doublemis1, 2022-07-19:
I can look at that also but, I can see that extended and doesn't match
./chip-tool networkcommissioning connect-network hex:**1161111128222222** 1 0 --Breadcrumb 2 - extended panid 1161111128222222

./chip-tool networkcommissioning add-or-update-thread-network hex:0e080000000000010000000300000f35060004001fffe00208**1111111128222222**0708fd5930b7d8b62f58051000112233445562778899aabbccddeeff030e4f70656e54687265746444656d6f010212340410005ed405199188e1a5c0dd607282d7480c0402a0fff8 22 0 - extended panid 1111111128222222

woody-apple, 2022-07-19:
Cert Blocker Review: It appears that we may need to change the test plan. @cecille can you please file an issue against the test plan and close this one?

Thanks!

Survensa, 2022-07-20:
> 

@doublemis1 Checked for extended panid but facing same issue :

1. `./chip-tool networkcommissioning connect-network hex:1111111107222222 48 0 --Breadcrumb 2`
2. `./chip-tool networkcommissioning add-or-update-thread-network hex:0e080000000000010000000300000f35060004001fffe0020811111111072222220708fd9957a92dadc7aa051000112233445966778899aabbccddeeff030f4f70656e54687265616444656d6f6a0102123404106ad73489ca556d19f8c47fd2150e58e30c0402a0fff8 48 0`

Full log : [cnet-4.12 error.txt](https://github.com/project-chip/connectedhomeip/files/9146905/cnet-4.12.error.txt)

doublemis1, 2022-07-21:
Hi, I was able to reproduce it.
The Border Router 1 and Border Router 2 are within the same network (not Thread Network) and may discover its services by DNS.  Thats way after connect DUT to the Border Router 2  network, the DUT informs that SRP services already existis (existing on Border Router 1).
```
I: 375759 [DL]SRP Client was stopped, because current server is no longer detected.
I: 375767 [DL]SRP Client was started, detected server: fd2f:a875:aab9:3927:a4b1:7250:5381:4135
D: 375776 [DL]OpenThread State Changed (Flags: 0x200002a4)
D: 375819 [DL]   Device Role: CHILD
D: 375822 [DL]   Partition Id: 0xD192944
D: 375831 [DL]OpenThread State Changed (Flags: 0x00000003)
D: 375837 [DL]   Thread Unicast Addresses:
D: 375881 [DL]        fd11:db::275:5101:6228:531d/64 valid preferred
D: 375887 [DL]        fd2f:a875:aab9:3927:0:ff:fe00:4c01/64 valid rloc
D: 375894 [DL]        fd2f:a875:aab9:3927:5a87:1ca3:6d30:7f7e/64 valid
D: 375901 [DL]        fe80::3481:909f:b6e4:e6d6/64 valid preferred
D: 375912 [DMG]ICR moving to [ Preparing]
D: 375916 [DMG]ICR moving to [AddingComm]
D: 375919 [DMG]ICR moving to [AddedComma]
D: 375925 [DMG]Endpoint 0, Cluster 0x0000_0030 update version to 192d259f 
D: 375931 [DMG]Decreasing reference count for CommandHandler, remaining 0 
I: 375940 [IN]Prepared secure message 0x20004ce4 to 0x000000000001B669 (1)  of type 0x9 and protocolId (0, 1) on exchange 12594r with MessageCounter:108659917.
I: 375954 [IN]Sending encrypted msg 0x20004ce4 with MessageCounter:108659917 to 0x000000000001B669 (1) at monotonic time: 000000000005BC92 msec
D: 375969 [DMG]ICR moving to [CommandSen]
D: 375973 [DMG]ICR moving to [AwaitingDe]
D: 375977 [DL]Using Thread extended MAC for hostname.
I: 375983 [DIS]Advertise operational node 529BA477D61319E8-0000000000000001
I: 375992 [SVR]Operational advertising enabled
I: 376146 [EM]Received message of type 0x10 with protocolId (0, 0) and MessageCounter:246502512 on exchange 12594r
D: 376156 [EM]Found matching exchange: 12594r, Delegate: (nil)
D: 376163 [EM]Rxd Ack; Removing MessageCounter:108659917 from Retrans Table on exchange 12594r
D: 376172 [EM]Removed CHIP MessageCounter:108659917 from RetransTable on exchange 12594r
I: 376181 [DL]OpenThread SED interval set to 1000ms
**E: 377284 [DL]SRP update error: domain name or RRset is duplicated**
```

I was able to pass this test when I depend only on docker networks by that way the border routers are on seperate networks.
Equipment:

- Linux PC
- 2x nrf52840dongle with RCP
- DUT

Creating docker networks:

- sudo docker network create --ipv6 --subnet fd11:db9:1::/64 -o com.docker.network.bridge.name=otbr20 otbr2
- sudo docker network create --ipv6 --subnet fd11:db8:1::/64 -o com.docker.network.bridge.name=otbr0 otbr
- sudo sysctl net.ipv6.conf.otbr0.accept_ra_rt_info_max_plen=128
- sudo sysctl net.ipv6.conf.otbr0.accept_ra=2
- sudo sysctl net.ipv6.conf.otbr20.accept_ra_rt_info_max_plen=128
- sudo sysctl net.ipv6.conf.otbr20.accept_ra=2

Run Border routers
- docker run -it --rm --privileged --network otbr -p 8080:80 --sysctl "net.ipv6.conf.all.disable_ipv6=0 net.ipv6.conf.all.forwarding=1" --name otbr --volume <RCP1 device path>:/dev/ttyACM0 <border router docker> --radio-url spinel+hdlc+uart:///dev/ttyACM0
- docker run -it --rm --privileged --network otbr2 -p 8081:80 --sysctl "net.ipv6.conf.all.disable_ipv6=0 net.ipv6.conf.all.forwarding=1" --name otbr_2 --volume <RCP2 device path>:/dev/ttyACM0 <border router docker> --radio-url spinel+hdlc+uart:///dev/ttyACM0
# Note web GUI for otbr1 will be avaliable on 0.0.0.0:8080, for otbr2 0.0.0.0:8081

Form networks:
Form networks on both Border Routers with different datasets.

Proceed the test.

doublemis1, 2022-07-22:
Currently, the SRP Replication feature is not supported on the current Thread 1.3, hence the border routers from two separate Thread Networks can discover its services within a local network (using backbone interface) but synchronization and replication handling are not implemented.

Survensa, 2022-08-01:
@doublemis1 I am able to connect to second network through second OTBR but the issue is I am not able to proceed further, I was forced to restart the whole setup.

Survensa, 2022-08-02:
@doublemis1 Once after the second network is connected, I am no able to execute any command as it end with Timeout error

Attached both chiptool and DUT log for referance :

- [cnet-4.12-cntl.txt](https://github.com/project-chip/connectedhomeip/files/9242735/cnet-4.12-cntl.txt)
- [cnet-4.12-dut.txt](https://github.com/project-chip/connectedhomeip/files/9242736/cnet-4.12-dut.txt)

Survensa, 2022-08-04:
Thanks @doublemis1 

Followed the steps to get connected and able to run the test case but not able to connect with first network once after disarming the Armfailsafe and ends with Timeout error 

Error log :

> I: 583601 [TS]Previous Last Known Good Time: 2022-07-29T05:46:32
> I: 583607 [TS]Reverted Last Known Good Time to previous value
> D: 583613 [DMG]Endpoint 0, Cluster 0x0000_0030 update version to 9380461a
> D: 583621 [ZCL]Failsafe timeout, tell platform driver to revert network credentials.
> E: 583629 [NP]Found Thread configuration backup: reverting configuration
> E: 583723 [DL]Long dispatch time: 256 ms, for event type 32782
> I: 583729 [FS]Fail-safe cleanly disarmed
> D: 583732 [DL]OpenThread State Changed (Flags: 0x11002046)
> D: 583738 [DL]   Device Role: DETACHED
> D: 583741 [DL]   Thread Unicast Addresses:
> D: 583745 [DL]        fdd0:6b61:12a4:6da6:38a8:de10:b0ba:7e89/64 valid
> D: 583751 [DL]        fe80::c831:832e:c849:8deb/64 valid preferred
> D: 583757 [DL]        fd11:92::a1b:f16d:d7f3:d2f2/64 valid preferred
> D: 583766 [DL]OpenThread State Changed (Flags: 0x10034000)
> D: 583771 [DL]   Network Name: OpenThreadDemo
> D: 583775 [DL]   PAN Id: 0x1234
> D: 583780 [DL]   Extended PAN Id: 0x2111111122222222
> D: 583785 [DL]   Channel: 11
> D: 583788 [DL]   Mesh Prefix: fdd0:6b61:12a4:6da6::/64
> I: 583796 [DL]CHIPoBLE advertising disabled because device is fully provisioned
> D: 583803 [DL]OpenThread State Changed (Flags: 0x1100107d)
> D: 583809 [DL]   Device Role: DETACHED
> D: 583812 [DL]   Thread Unicast Addresses:
> D: 583816 [DL]        fdd0:6b61:12a4:6da6:38a8:de10:b0ba:7e89/64 valid
> D: 583822 [DL]        fe80::c831:832e:c849:8deb/64 valid preferred
> D: 583828 [DL]        fd11:92::a1b:f16d:d7f3:d2f2/64 valid preferred
> I: 584675 [DL]SRP Client was stopped, because current server is no longer detected.
> I: 584682 [DL]SRP Client was started, detected server: fdd0:6b61:12a4:6da6:3720:f78e:b659:724e
> D: 584692 [DL]OpenThread State Changed (Flags: 0x200002a4)
> D: 584734 [DL]   Device Role: CHILD
> D: 584737 [DL]   Partition Id: 0x17775288
> D: 584746 [DL]OpenThread State Changed (Flags: 0x00000003)
> D: 584752 [DL]   Thread Unicast Addresses:
> D: 584796 [DL]        fd11:13::a8e0:a386:f9a4:fde0/64 valid preferred
> D: 584804 [DL]        fdd0:6b61:12a4:6da6:0:ff:fe00:1003/64 valid rloc
> D: 584810 [DL]        fdd0:6b61:12a4:6da6:38a8:de10:b0ba:7e89/64 valid
> D: 584817 [DL]        fe80::c831:832e:c849:8deb/64 valid preferred
> D: 585777 [DL]SRP update succeeded

bzbarsky-apple, 2022-08-08:
@cecille After a ConnectNetwork, the test plan expects to still receive the response to the original command on the old network that the device is no longer connected to.

doublemis1, 2022-08-09:
@bzbarsky-apple, currently (at least nrfconnect platform) transmits a response to the old network, and right after sending the response DUT, switches to the new network.

cecille, 2022-08-09:
Connect network command is sent in steps 7 and 16. Neither one of those steps has any expected outcome specified. I suppose it might have been better to add an explicit note that no response is required, but I don't think it's wrong.

bzbarsky-apple, 2022-08-09:
The stale SRP record thing sounds like https://github.com/project-chip/connectedhomeip/issues/21101

Survensa, 2024-02-16:
Issue fixed in this PR: https://github.com/project-chip/connectedhomeip/pull/32156

cjandhyala, 2025-09-22:
closing the issue per Pradeep's comments

closed 2025-09-22 by cjandhyala