project-chip/connectedhomeip #20809: Audit matter-devices.xml for sanity

bzbarsky-apple, 2022-07-15:
#### Problem
We have all sorts of bad copy/paste bugs in matter-devices.xml.  For example, On/Off Plug-In unit has:
```
            <include cluster="Binding" client="true" server="false" clientLocked="false" serverLocked="true">
                <requireAttribute>BINDING</requireAttribute>
            </include>
```
which defaults to including a binding client for some reason, even though the device type does not mention that.

 #### Proposed Solution
Go through this file, align with spec.

woody-apple, 2022-08-17:
This may be resolved with a bunch of PRs like so:
https://github.com/project-chip/connectedhomeip/pull/21911
Can you confirm?

yufengwangca, 2022-08-22:
Is Binding cluster one of Base clusters?

Base Cluster Requirements for Matter
Each device type definition SHALL include these clusters, as a minimum set, based on the conformance defined below. This conformance table SHALL assume the Matter conformance condition is TRUE (in Conformance column).
Descriptor
Binding 
FixedLabel
UserLabel

Base Cluster Requirements for Zigbee
Each device type definition SHALL include these clusters, as a minimum set, based on the conformance defined below. This conformance table SHALL assume the Zigbee conformance condition is TRUE (in Conformance column).
Basic

yufengwangca, 2022-08-22:
Shall those five base clusters be included to each Device Type by default? The device type don't need to mention them explicitly.

yufengwangca, 2022-08-23:
Each device type is attached to an endpoint, same device type could be attached to different endpoints, so we should have each device type definition include Descriptor cluster which is mandatory base cluster

yufengwangca, 2022-08-23:
Binding cluster is defined as an Optional base cluster on the server side, so we should be turned off AND user can not turn it on at the client and is turned off but user CAN turn it on at the server side.

<include cluster="Binding" client="false" server="false" clientLocked="true" serverLocked="false">
           <requireAttribute>BINDING</requireAttribute>
</include

PR #22105 (2022-08-24): Include missing Descriptor cluster in Secondary Network Commissioning…
… Device Type

#### Problem
What is being fixed?  Examples:
* According spec, we have five base clusters defined for Matter
```
Base Cluster Requirements for Matter
Each device type definition SHALL include these clusters, as a minimum set, based on the conformance defined below. This conformance table SHALL assume the Matter conformance condition is TRUE (in Conformance column).
Descriptor
Binding
FixedLabel
UserLabel
```
Among those 5 Matter base clusters, Descriptor is mandatory and we need to include this cluster for ea [... cut]
PR #22119 (2022-08-25): Update Binding cluster requirement in device type to align with the spec
#### Problem
What is being fixed?  Examples:
* Binding cluster is defined as one of optional base clusters on the server side,  each device type definition SHALL include this cluster without explicitly specification. I saw some device types have included this cluster, but not all of them conform to the Binding cluster requirement in the spec.

https://github.com/CHIP-Specifications/connectedhomeip-spec/blob/master/src/device_types/BaseDeviceType.adoc  

* Binding cluster is defined as Server side cluster, and it is not ma [... cut]
closed 2022-08-24 by yufengwangca