project-chip/connectedhomeip #20810: ACL - audit CAT validation

andy31415, 2022-07-15:
#### Problem
Side-discussion regarding ACL validation in 
https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/cluster/AccessControl.adoc#tc-acl-2-4-acl-attribute resulted in us wondering about how subjects are being validated.

Specifically we are testing now for some invalid values like 0xFFFF_FFFF_FFFF_FFFF, 0xFFFF_FFFF_FFFF_0000 and 0xFFFF_FFFD_0000_0000 however additional review would be required.

 #### Proposed Solution
@tcarmelveilleux volunteered to be assigned an `Audit ACL validation of CATs` so here is the issue for that.

tcarmelveilleux, 2022-07-26:
For collisions of CAT versions in subjects, they are independant subjects. We already validate for NOCs that there are no collisions of CAT identifier versions. However, for ACL enforcement, I don't think we should internally audit for complex "possibly ugly" policies. If there are multiple versions of a single CAT identifier present in `Subjects` of an ACL, it means *an admin* decided to have it there. The ACL privilege algorithm tries to match the highest privielege, skipping all non-matching. Therefore, the smaller version of a CAT identifier will eventually match, if a CAT of the incoming subject has that version or later. This is OK and we should not error-out on it. Otherwise we need complex  new code in ACL cluster to guard against what amounts to just a "weird" (but not illegal) configuration. In Access control enforcement, *all subjects are considered independently*.

I wouild argue that after finding `version 0` CATs are properly disallowed, and that CAT collisions are disallowed in NOCs, we don't need to audit the subjects of ACL entries for CATs beyond that.

bzbarsky-apple, 2022-07-26:
> I wouild argue that after finding version 0 CATs are properly disallowed, and that CAT collisions are disallowed in NOCs, we don't need to audit the subjects of ACL entries for CATs beyond that.

I agree.

closed 2022-07-26 by tcarmelveilleux