project-chip/connectedhomeip #20838: Test Case Timeout for CI execution is causing failure for TC-CADMIN(MF) test cases

kvikrambhat, 2022-07-16:
#### Problem
Test case timeout of 120 seconds specified for running tests on CI is cashing TC-CADMIN(MF) test cases to crash. 

 #### Proposed Solution
Increase timeout to 200 seconds to accommodate TC-CADMIN(MF) test cases

[image]

bzbarsky-apple, 2022-07-18:
@woody-apple @manjunath-grl @kvikrambhat The point is, tests that take longer than about 2 minutes to run _should not be running in CI_.  They should be marked as "manual" tests.  The whole point here was to avoid people adding new very laggy tests to CI.

Why was the timeout raised in #20821?

bzbarsky-apple, 2022-07-18:
We should not be raising this timeout.  Multi-minute tests should not be running automatically in CI.

closed 2022-07-18 by bzbarsky-apple

kvikrambhat, 2022-07-19:
@bzbarsky-apple We took the call to increase the timeout to 3 minutes to accommodate the MF test scripts, We took the call as we a few PRs lined up for SVE, we will disable these test and reduce the timeout in the next PR.

raju-apple, 2022-07-19:
@kvikrambhat Lets move that test case to manual list for CI. It will still be available for execution in TH since all the test steps are not disabled.

bzbarsky-apple, 2022-07-20:
> we will disable these test and reduce the timeout in the next PR

I already did that in https://github.com/project-chip/connectedhomeip/pull/20870

kvikrambhat, 2022-07-20:
Thanks you,will make sure we dont increase it going forward.