project-chip/connectedhomeip #20844: IdentifyType wrong value on boot of DUT

innovation-matters-iot, 2022-07-16:
#### Problem
A value of 0x01 for IdentifyType on the identify cluster:

0x01 | Light output | Light output of a lighting product.

 set by ZAP tool is read as value 0x02 by the DUT on boot.
[image]
[image]
[image]

bzbarsky-apple, 2022-07-18:
The ZAP default is not used for identify in practice.  What matters is how the `Identify` is constructed.  How is it constructed in the case of this DUT @innovation-matters-iot ?

innovation-matters-iot, 2022-07-19:
It was constructed the wrong way. Thanks for the help!

closed 2022-07-19 by innovation-matters-iot