project-chip/connectedhomeip #20851: TC-BINFO-2.1 Issue with writing '[{ 4 , 4 }]' value to Attributes

AlvinHsiao, 2022-07-18:
#### Problem
It should be possible to run this test in the same way as in the test plan, but chip-tool does not accept write-by-id command with this value as the second argument:

 * actual behavior
 ubuntu@ubuntu:~/apps$ ./chip-tool basic write-by-id 0x0013 '[{ 4 , 4 }]' 10 0
[1658127024.709694][15427:15427] CHIP:TOO: Error parsing JSON for attribute-values:
[1658127024.710508][15427:15427] CHIP:TOO:     Missing '}' or object member name
[1658127024.711011][15427:15427] CHIP:TOO: [{ 4 
[1658127024.711548][15427:15427] CHIP:TOO: ---^
 * steps to reproduce
 ./chip-tool basic write-by-id 0x0013 '[{ 4 , 4 }]' 10 0
 *  system configuration
Chip-Tool: RPI
DUT used : Infineon CYW30739
app use: Lighting-app
chip-tool and DUT commit id : https://github.com/project-chip/connectedhomeip/commit/333fa10187d34ed1209cddd6c1fe40d55968a93a (sve SHA)

 * logs
[TC-BINFO-2.1.txt](https://github.com/project-chip/connectedhomeip/files/9130019/TC-BINFO-2.1.txt)

bzbarsky-apple, 2022-07-18:
> chip-tool basic write-by-id 0x0013 '[{ 4 , 4 }]' 10 0

Where is this line coming from?  It's totally wrong.  What is it trying to do? @AlvinHsiao 

@cjandhyala

bzbarsky-apple, 2022-07-18:
I assume this is this test step:

> TH writes the CapabilityMinima value with CaseSessionsPerFabric set to 4 and SubscriptionsPerFabric set to 4

If so, the correspoding command would be:
```
chip-tool basic write-by-id 0x0013 '[{ "0": 4 , "1": 4 }]' 10 0
```

AlvinHsiao, 2022-07-19:
@bzbarsky-apple It's from step 57 of TC-BINFO-2.1 of SVE1_TestPlanVerificationSteps_V1.2.

bzbarsky-apple, 2022-07-19:
@AlvinHsiao Please file an issue on the test plan verification steps.

AlvinHsiao, 2022-07-19:
Found it's fixed in SVE1_TestPlanVerificationSteps_V1.3 with aa9457e6b94b735076dff6297176183bf9780177.

cjandhyala, 2022-07-19:
closing the issue per @AlvinHsiao comments that it is fixed

closed 2022-07-19 by cjandhyala