project-chip/connectedhomeip #20862: UDP endpoint bound to wrong interface when sending group messages

gjc13, 2022-07-18:
#### Problem

When testing group messages in SVE, we found that chip-tool fails to send group multicast messages.

Test steps:

```
./chip-tool accesscontrol write acl '[{"fabricIndex": 1, "privilege": 5, "authMode": 2, "subjects": [112233], "targets": null },{"fabricIndex": 1, "privilege": 4, "authMode": 3, "subjects": [1], "targets": null }]' 1 0

./chip-tool groupkeymanagement key-set-write '{"groupKeySetID": 42, "groupKeySecurityPolicy": 0, "epochKey0": "d0d1d2d3d4d5d6d7d8d9dadbdcdddedf", "epochStartTime0": 2220000,"epochKey1": "d1d1d2d3d4d5d6d7d8d9dadbdcdddedf", "epochStartTime1": 2220001,"epochKey2": "d2d1d2d3d4d5d6d7d8d9dadbdcdddedf", "epochStartTime2": 2220002 }' 1 0

./chip-tool groups add-group 0x0001 grp1 1 1

./chip-tool groupkeymanagement write group-key-map '[{"groupId": 1, "groupKeySetID": 42, "fabricIndex": 1}]' 1 0

./chip-tool groupsettings add-group grp1 0x0001

./chip-tool groupsettings add-keysets 0x0042 0 0x000000000021dfe0 hex:d0d1d2d3d4d5d6d7d8d9dadbdcdddedf

./chip-tool groupsettings bind-keyset 0x0001 0x0042

./chip-tool onoff toggle 0xffffffffffff0001 1
```

The multicast message was not sent after executing the `chip-tool onoff toggle` command.

The bug is caused by the UDP endpoint bound to an incorrect interface. The `IPv6Bind` function is called with a null interface during initialization and it turned out that multicast messages won't be set under this scenario.

I tried to force the interface with the following patch:

```diff
diff --git a/src/inet/UDPEndPointImplSockets.cpp b/src/inet/UDPEndPointImplSockets.cpp
index 93b0828f5..59b91e762 100644
--- a/src/inet/UDPEndPointImplSockets.cpp
+++ b/src/inet/UDPEndPointImplSockets.cpp
@@ -102,6 +102,11 @@ CHIP_ERROR IPv6Bind(int socket, const IPAddress & address, uint16_t port, Interf
     else
     {
 #ifdef IPV6_MULTICAST_IF
+        ChipLogProgress(Inet, "IPV6_MULTICAST_IF interface %d", interfaceId);
+        if (interfaceId == 0)
+        {
+            interfaceId = 2;
+        }
```
The patch will enable group commands on my ethernet link.

 #### Proposed Solution

Explicitly bind to an interface when sending multicast packets.

gjc13, 2022-07-18:
@jepenven-silabs Would you help to look at this issue?

bzbarsky-apple, 2022-07-18:
@gjc13 Are you running chip-tool on Mac or Linux?

gjc13, 2022-07-19:
> 

On Linux

gjc13, 2022-07-19:
> @gjc13 Are you running chip-tool on Mac or Linux?

On Linux

woody-apple, 2022-07-20:
Spec Review: Assigning to @jepenven-silabs to take a look. thanks!

jepenven-silabs, 2022-07-20:
@gjc13 @woody-apple I'll take a look, but this seems really platform/OS specific. I'll try to reproduce on my setup, but should I fail I suggest that we come up with a generic way to subscribe to a multicast interface. 

Something along the line of : If an interface is not specified by the application layer, subscribe to all available interface.

@bzbarsky-apple you also ran into this on your side no ?

bzbarsky-apple, 2022-07-20:
Yes, Darwin had some issues here too.  I am not sure what the status there is.

sethunk, 2022-07-22:
@gjc13 Kindly swap step-3 and step-4 so that the group can be added successfully otherwise the response for addgroup will be UNSUPPORTED_ACCESS.

Have attached the logs by executing the above commands in the following order to send multicast messages.

> ./chip-tool accesscontrol write acl '[{"fabricIndex": 1, "privilege": 5, "authMode": 2, "subjects": [112233], "targets": null },{"fabricIndex": 1, "privilege": 4, "authMode": 3, "subjects": [1], "targets": null }]' 1 0
> 
> ./chip-tool groupkeymanagement key-set-write '{"groupKeySetID": 42, "groupKeySecurityPolicy": 0, "epochKey0": "d0d1d2d3d4d5d6d7d8d9dadbdcdddedf", "epochStartTime0": 2220000,"epochKey1": "d1d1d2d3d4d5d6d7d8d9dadbdcdddedf", "epochStartTime1": 2220001,"epochKey2": "d2d1d2d3d4d5d6d7d8d9dadbdcdddedf", "epochStartTime2": 2220002 }' 1 0
>  
> ./chip-tool groupkeymanagement write group-key-map '[{"groupId": 1, "groupKeySetID": 42, "fabricIndex": 1}]' 1 0
> 
>   ./chip-tool groups add-group 0x0001 grp1 1 1 
> 
>  ./chip-tool groupsettings add-group grp1 0x0001
> 
> ./chip-tool groupsettings add-keysets 0x0042 0 0x000000000021dfe0 hex:d0d1d2d3d4d5d6d7d8d9dadbdcdddedf
> 
> ./chip-tool groupsettings bind-keyset 0x0001 0x0042
> 
> ./chip-tool onoff toggle 0xffffffffffff0001 1

jadhavrohit924, 2022-07-26:
@sethunk  I followed exact same steps mentioned by you, still failing to multicast `chiptool toggle command.
Running chiptool on mac.

[Group_Communication_Device_logs.txt](https://github.com/project-chip/connectedhomeip/files/9190096/Group_Communication_Device_logs.txt)
[Group_Communication.txt](https://github.com/project-chip/connectedhomeip/files/9190098/Group_Communication.txt)
`

bzbarsky-apple, 2022-07-26:
@jadhavrohit924 On Mac, group communication does not work properly at the moment.

jepenven-silabs, 2022-07-27:
I'm able to reproduce on my linux machine by only if the WiFi adapter is the sole interface available. Otherwise everything is working fine on both my Mac and Linux with ethernet. 

Can any of you @bzbarsky-apple @jadhavrohit924 @sethunk @gjc13 produce wiresharks logs? the only way to be sure that chip-tool isn't sending a multicast packet is actually by sniffing the network for it. 

you can use this filter `ipv6.addr >= ff05::` to reduce network noise and filter out Matter multicast IPV6 packets

jepenven-silabs, 2022-07-27:
More over, to save time, you can use this command directly after compiling `chip-tool` without having to setup groups keys or group 

`./out/debug/chip-tool onoff on 0xffffffffffff0101 1`

chip-tool actually already has two hardcoded group used for test like this. You should be able to see a packet with wireshark if everything is working on the chip-tool side (which is the case with my setup and master from today).

jepenven-silabs, 2022-07-27:
So after investigation, I came to the conclusion that while this may appear as a bug it's not really a bug.

Explanation : 
In the socket implementation the default interface value (0) is used in many places. This usually works for standard communication including multicast. The default interface is set by the operating system and may vary from computer to computer.

Now the problem that we are facing is as follow (from my understanding) Either the default interface used to send multicast command doesn't actually support multicast, doesn't have an IPV6 address or simply is not connected. One of these might be true for everyone setup except for those for whom multicast actually works.

So the possibles solutions are 

1. The App layer specify which interface to use instead of using the default interface value. This can easily be done by calling `SetInterfaceId(desired interface)` in `server.cpp` or anywhere a UDP endpoint is created. 

2. Somewhere in the creation of the message we set the value for the interface we wish to send it to. This can be done by setting the `Interface ` field of the `IPPacketInfo` object.  

3. Inside the UDP endpoint, add calls to actually verify is the interface is present or not. Currently the `isPresent()` method is a big joke since it compares the value of the interface against the `kPlatformNull` which is zero. Meaning that by default an endpoint using the "default" network interface will fail ALL `isPresent()' check. Moreover, checking not only if the interface is really present, but if the interface is actually up (connected) would be the best. And then if the interface isn't able to send the multicast message, we switch to another interface with hopes that we managed to send it to the right one at some point.  

Solution 1 is the easiest to implement but it has the dreadful drawback of actually knowing in advance the interfaceId to use, or to find a way to pass it to chip-tool at run time (yet another parameter)

Solution 3 would be the most reliable, meaning that should the ethernet cable of the pc get disconnected, chip-tool would still be able to send multicast commands through the WiFi adapter without any issue. But this open the door for an UDPEndpoint to send messages through an interface that it wasn't bound with in the first place. Which might be acceptable if and only if it was initialized with the default interface. But the second problem is as follow : How do we know to which interface to send the multicast packet ? Do we add more OS ish specific functions (ioctl) to find the next up and running interface and send it blindly ? do we send it to all up and running interfaces ?

Roping in @turon 
@bzbarsky-apple

turon, 2022-07-27:
Using the default interface but letting the user / application explicitly specify the interface (option 1) seems like the best approach to me.  Having an option to "bind to all interfaces" (option 3?) seems like a nice feature as well, but perhaps not as critical as being able to explicitly selecting at least one working interface (option 1).

jepenven-silabs, 2022-07-27:
The problem with 1 is, if implemented, it would required changing ALL tests related to groups to specify an interface, on top of adding steps to find which interface is the right one....

turon, 2022-07-27:
How about this proposal:

* Add a platform interface to build a list of all Link Local (LL) interfaces [fe80::X]%interface
* Default to using this list of all LL interfaces for multicast
* Provide an API hook to let the app layer filter that list or replace it with an explicitly constructed list of LL interfaces it wants to use for group comms.

A per-message API seems like over-kill for this.  The typical flow would be for a device to setup the list of interfaces it wants to use and then group comms send and receive across that list.

Now whether we should support multi-honing, where each Fabric can have an independent list of LL interfaces for multicast, I think we can defer to post-1.0.

The impact on the test harness for such a scheme would ideally be a few additional calls in the platform initialization code, and then some additional logic in the multicast send code to iterate through the interfaces.

For Thread / WiFi devices, sending on all is what you want.
For WiFi / Ethernet devices, the question of spam is an open one, yes.  Worst case each message gets sent twice on the common LAN.

jepenven-silabs, 2022-07-29:
Removing the spec label as this issue is a setup issue (OS configuration issue).

PR #21351 (2022-07-29): [Inet][Transport] Multicast Multihoming : Socket implementation
#### Problem
Fixes #20862 

#### Change overview
Send Multicast packet on all available interface.

An available interface is one which has an IPV6 link local address (`fe80::`) and that supports multicast

This change should only affect the Darwin and Linux build. 
Zephyr build shouldn't affected by this.
OpenThread and LwIP Implementation shouldn't be affected by this either. 

#### Testing
Tested with chip-tool + wireshark on both Linux (ubuntu 20.04) and MacOS
closed 2022-07-29 by woody-apple