project-chip/connectedhomeip #20879: [TC-LUNIT-3.1]When TH writes TemperatureUnit attribute value more than 2, DUT responds as success instead CONSTRAINT_ERROR.

Rajashreekalmane, 2022-07-18:
**Test Case ID**
[TC-LUNIT-3.1] Read and Write Unit Localization Cluster Attributes [{DUT_Server}]

**Describe the issue:-**
As per test-plan test step 8 is expecting to write 5 to TemperatureUnit attribute  from TH to DUT,  DUT should responds with constraint error But currently it is responding as success.

 **expected behavior**:-CONSTRAINT_ERROR(0x87)
 * **actual behavior**:-Success(0x00)
 * **steps to reproduce**:-./chip-tool unitlocalization write temperature-unit 5 1 0
 * **system configuration**
Chip-tool - RPI-4, 8GB RAM
DUT - RPI - RPI-4, 8GB RAM
Platform : IP and wifi
Commit-ID:0390a8d33b1d1c709fcc4499eab6e132428ad5e2
Dut application: all-clusters-app

**Test-plan Reference:-**
https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/cluster/unitlocalization.adoc#tc-lunit-3-1-read-and-write-unit-localization-cluster-attributes-dut_server

**Spec Reference:-**
https://github.com/CHIP-Specifications/connectedhomeip-spec/blob/master/src/service_device_management/LocalizationUnit.adoc#data-types

**PFA logs:-**
[[TC-LUNIT-3.1].txt](https://github.com/project-chip/connectedhomeip/files/9134257/TC-LUNIT-3.1.txt)

Rajashreekalmane, 2022-07-19:
For TemperatureUnit attribute should support only 0, 1, and 2 but currently If we try to write 3 to 255, the response is Success

andy31415, 2022-07-20:
Seems very similar with CONSTRAINT_ERROR that we had to return for ACLs.

Taking this for now to see if we can get some generic solution (I would like to convert any INVALID_ARGUMENT into CONSTRAINT_ERROR).

I will also have to implement validation for the temperature values, I believe those will have to return INVALID_ARGUMENT interally.

andy31415, 2022-07-20:
It seems these unit tests were not automated. I will automate this one for now.

andy31415, 2022-07-20:
I automated 3_1 yaml fully.

It turns out this is not running CHIP_ERROR, so no centralized solution was needed for this specifically.

PR #21003 (2022-07-21): Unit localization cluster constraints
#### Problem

Unit localization cluster does not validate the temperature unit.
Fixes #20879 

#### Change overview
- set min/max values for the unit localization
- add unit tests

#### Testing
CI should run tests. Also ran locally:

```
./scripts/tests/run_test_suite.py --target Test_TC_LUNIT_3_1 --chip-tool ./out/linux-x64-chip-tool/chip-tool run --all-clusters-app ./out/linux-x64-all-clusters/chip-all-clusters-app --tv-app none --lock-app none --ota-provider-app none --ota-requestor-app none
```
PR #21080 (2022-07-21): Unit localization cluster constraints
#### Problem

Unit localization cluster does not validate the temperature unit.
Fixes #20879 

#### Change overview
- set min/max values for the unit localization
- add unit tests

#### Testing
CI should run tests. Also ran locally:

```
./scripts/tests/run_test_suite.py --target Test_TC_LUNIT_3_1 --chip-tool ./out/linux-x64-chip-tool/chip-tool run --all-clusters-app ./out/linux-x64-all-clusters/chip-all-clusters-app --tv-app none --lock-app none --ota-provider-app none --ota-requestor-app none
```
closed 2022-07-21 by woody-apple