project-chip/connectedhomeip #20882: AdministratorCommissioning cluster not implemented per spec

bzbarsky-apple, 2022-07-18:
#### Problem
Several problems:
1. WindowStatus does not return WindowNotOpen during initial commissioning, as far as I can tell.  It should.
2. AdminFabricIndex and AdminVendorId don't return null when they should (i.e. any time status is WindowNotOpen).

Depending on how https://github.com/CHIP-Specifications/connectedhomeip-spec/issues/5473 is resolved that could be an issue too.

 #### Proposed Solution
Add tests, align with spec.

tcarmelveilleux, 2022-07-27:
https://github.com/CHIP-Specifications/connectedhomeip-spec/pull/5474 will fix the behavior in spec, but then will need to follow in SDK

woody-apple, 2022-07-29:
Cert Blocker Review: Marking this as 'needs provisional' unless we have a resolution to this issue today.

woody-apple, 2022-07-29:
Cert Blocker Review: Removing cert blocker, given we don't know of any tests at this point that are failing.

closed 2022-07-31 by woody-apple