project-chip/connectedhomeip #20885: [TC-BIND-2.1] -   Binding entries for Endpoint 2 still showing the success response after removing the entries

somu1710, 2022-07-18:
**Description:**

- In Binding cluster test case [TC-BIND-2.1] - writing binding entries to the two endpoints.**(Ex:Endpoint-1 and 2)**..Then,send the on command and checking the on/off value for both endpoints and got success.
- But After removing the binding entry for endpoint 2 ,send off command to both entries .Both endpoints receiving the success response and on/off attribute shows "FALSE" value.

**Expected outcome:**
After removing the binding entry for `Endpoint 2 `,send off command to both entries .Endpoint 2 should not receive the off command from TH1 and the `ON/OFF` attribute should show the value as `"TRUE" .`

**Actual outcome:**
 After removing the binding entry for endpoint 2 ,send off command to both entries .Both endpoints  has receiving the success response and on/off attribute shows "FALSE" value.

DUT-Raspberrypi
APP used - All-clusters-app

**Draft Doc link:**
https://docs.google.com/spreadsheets/d/1h4d4J90KAxFqUL_cXZHyM3pHjHCEXzFITkQhN2KMqDY/edit?pli=1#gid=608563054

**Test plan link:**
https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/cluster/binding.adoc#321-tc-bind-21-binding-cluster-attributes-node-dut-controller

**Logs for reference:**
[TC-BIND-2.1.txt](https://github.com/project-chip/connectedhomeip/files/9134547/TC-BIND-2.1.txt)
[TC-BIND-2.1 -- 2ND.txt](https://github.com/project-chip/connectedhomeip/files/9134550/TC-BIND-2.1.--.2ND.txt)

woody-apple, 2022-07-19:
Cert Blocker Review: Assigning cert blocker.

bzbarsky-apple, 2022-07-19:
So looking at the [TC-BIND-2.1.txt](https://github.com/project-chip/connectedhomeip/files/9134547/TC-BIND-2.1.txt) log, I am pretty confused.  It's doing the following, via chip-tool:

1. Commission a device.
2. Open a commissioning window.
3. Write the binding attribute.
4. Send "On" command to endpoint 1.
5. Send "On" command to endpoint 2.
6. Read OnOff attribute on endpoint 1.
7. Write binding attribute.
8. Send "Off" command to endpoint 1.
9. Send "Off" command to endpoint 2.
10. Read OnOff attribute on endpoint 1.
11. Read OnOff attribute on endpoint 2.

These steps have nothing to do with the test plan for TC-BIND-2.1 as far as I can see.  In TC-BIND-2.1 the entity sending the "On" and "Off" commands is the same as the DUT that has its bindings written, which is just not the case here.

@cjandhyala @somu1710 The test steps here need to be fixed.

bzbarsky-apple, 2022-07-19:
OK, looking at the verification step spreadsheet linked above at https://docs.google.com/spreadsheets/d/1h4d4J90KAxFqUL_cXZHyM3pHjHCEXzFITkQhN2KMqDY/edit?pli=1#gid=608563054 (thank you for the link!), the verification steps are just wrong.  Filed https://github.com/CHIP-Specifications/chip-test-plans/issues/1831