project-chip/connectedhomeip #20927: Is there a plan to support arm64 simulator on m1 device?

zzz2016, 2022-07-19:
I'm going tu build the CHIP framework for arm64 simulator, but it failed with message:
`ld: in /Users/zzz/Library/Developer/Xcode/DerivedData/CHIP-gdbowhebkylyoweuzksbppqwqxiy/Build/Intermediates.noindex/ArchiveIntermediates/CHIP Framework/IntermediateBuildFilesPath/CHIP.build/Release-iphonesimulator/CHIP.build/out/lib/libCHIP.a(libChipController.CHIPCluster.cpp.o), building for iOS Simulator, but linking in object file built for iOS, for architecture arm64`

Here is my xcodebuild command:
x```
codebuild archive \
-scheme CHIP\ Framework \
-archivePath ../builds/arm64-simulator.xcarchive \
-sdk iphonesimulator \
-arch arm64
```
Is it my fault or CHIP.framework does't supprot the arm64 simulator build?

bzbarsky-apple, 2022-07-19:
@woody-apple @jtung-apple

bzbarsky-apple, 2023-02-17:
@zzz2016 Is this still a problem?  M1 support has been improved quite a bit....

woody-apple, 2023-06-23:
Please re-open if this is still an issue