project-chip/connectedhomeip #20930: [TC-DRLK-2.9] Door-lock Set Credential Command is returning User Index as NULL.

Ashwinigrl, 2022-07-19:
**1.Problem** -In TC-DRLK-2.9,  in Set Credential Command the value for the UserIndex is provided as 1 and the response is success but in the Set Credential Response the value of UserIndex is returned as NULL.

 **2.steps to reproduce**
./chip-tool doorlock set-user 0 1 xxx 6452 1 0 0 1 1  --timedInteractionTimeoutMs 1000

 ./chip-tool doorlock set-credential 0 '{ "credentialType" : 1 , "credentialIndex" : 1 }' 123456 1 0 0 1 1 --timedInteractionTimeoutMs 1000

> [1655373390.829689][2679:2684] CHIP:TOO:   SetCredentialResponse: {
> [1655373390.829737][2679:2684] CHIP:TOO:     status: 0
> [1655373390.829854][2679:2684] CHIP:TOO:     userIndex: null
> [1655373390.829892][2679:2684] CHIP:TOO:     nextCredentialIndex: 2
> [1655373390.829926][2679:2684] CHIP:TOO:    }
> 

**3.Test Environment**
App Used - lock app
Platform - Chip-tool - RPI-4, 8GB RAM
DUT -  RPI-4, 8GB RAM
Network - Ethernet
commit id - 742b80e93346926435743976271ebf1362a3a85c
 
**4.logs**
[Set Credential cmd logs.txt](https://github.com/project-chip/connectedhomeip/files/9141353/Set.Credential.cmd.logs.txt)

bzbarsky-apple, 2022-07-19:
@Morozov-5F

woody-apple, 2022-07-19:
Cert Blocker Review: Assigning to @Morozov-5F

mykrupp, 2022-07-25:
I will take on this one.

bzbarsky-apple, 2022-07-28:
This looks like the right behavior per spec, insofar as the spec makes any sense (see https://github.com/CHIP-Specifications/connectedhomeip-spec/issues/5507).

In particular, the commands above add a credential _but use an existing user for the credential_, so the userIndex in the response is null because the "Set Credential" did not create a user.

woody-apple, 2022-07-29:
Cert Blocker Review: @cjandhyala to follow up on the test here.

closed 2022-07-29 by woody-apple