project-chip/connectedhomeip #20956: What happen with next version of Matter data model that want to add something stated 6 ?

woody-apple, 2022-07-19:
What happen with next version of Matter data model that want to add something stated 6 ?
same question apply  for all these enum.
When we iterate and add new feature or bump up enum list , then this might be an issue 
I just wondering how this gonna be handled in the future

_Originally posted by @jmeg-sfy in https://github.com/project-chip/connectedhomeip/pull/20937#discussion_r924770608_

woody-apple, 2022-07-19:
Note, filed this @jmeg-sfy

bzbarsky-apple, 2022-07-20:
See https://github.com/project-chip/connectedhomeip/pull/20937#discussion_r924979917

@mrjerryjohns and I don't think there's a problem here.

bzbarsky-apple, 2023-01-17:
Note also discussion in https://github.com/project-chip/connectedhomeip/issues/24368

closed 2023-10-15 by stale[bot]