project-chip/connectedhomeip #20976: nordic 平台light-switch-app&lighting-app  Demo待机一晚均会出现死机现象

ZhongShangWen, 2022-07-20:
**light-switch-app log**

[image]

**lighting-app log**

[image]

ZhongShangWen, 2022-07-20:
**chip commit 记录**
[image]

SteveCruise, 2022-07-21:
兄弟加油，30w智能家居装修就等matter成熟了

Damian-Nordic, 2022-08-01:
I think it has been fixed in the light switch sample and later, more generally by:
> commit 72ada957a02a6056499779f6184b10a8252431b9
> Author: Damian Królik <66667989+Damian-Nordic@users.noreply.github.com>
> Date:   Fri Jul 1 23:29:22 2022 +0200
>
> [zephyr] Fix crash on saving TotalOperatingHours to KVS (#20214)

Could you confirm that the issue is not appearing using a revision newer than that?

closed 2023-10-15 by stale[bot]