project-chip/connectedhomeip #20979: [SVE1] Groups not working on latest master

chirag-silabs, 2022-07-20:
#### Problem
Groups is not working on the latest master.  

 * expected behavior
After adding a DUT into a group sending the group to group id changes the state of DUT as well 

 * actual behavior
No change on the DUT side

 * steps to reproduce
./chip-tool groups add-group 0x0001 grp1 1122 1
./chip-tool groupkeymanagement key-set-write ''{""groupKeySetID"": ""0x42"",""groupKeySecurityPolicy"": 0, ""epochKey0"":""d0d1d2d3d4d5d6d7d8d9dadbdcdddedf"", ""epochStartTime0"": 2220000,""epochKey1"":""d1d1d2d3d4d5d6d7d8d9dadbdcdddedf"", ""epochStartTime1"": 2220001,""epochKey2"":""d2d1d2d3d4d5d6d7d8d9dadbdcdddedf"", ""epochStartTime2"": 2220002 }'' 1122 0
./chip-tool groupkeymanagement write group-key-map "'[{""groupId"": 1, ""groupKeySetID"": ""0x42"", ""fabricIndex"": 1}]'" 1122 0
./chip-tool accesscontrol write acl "'[{""fabricIndex"": 1, ""privilege"": 5, ""authMode"": 2, ""subjects"": [112233], ""targets"": null },{""fabricIndex"": 1, ""privilege"": 3, ""authMode"": 3, ""subjects"": [1], ""targets"": null }]'" 1122 0

chip-tool
./chip-tool groupsettings add-group grp1 0x0001
./chip-tool groupsettings add-keysets 0x42 0 0x000000000021dfe hex:d0d1d2d3d4d5d6d7d8d9dadbdcdddedf
./chip-tool groupsettings bind-keyset 0x0001 0x42

./chip-tool onoff on 0xffffffffffff0001 1

 * system configuration
DUT: EFR32 + RS9116

 #### Proposed Solution
It is working fine on the commit (https://github.com/project-chip/connectedhomeip/pull/20136) and fails on the latest SVE. Maybe one commit is affecting the functionality of groups.

rosahay-silabs, 2022-07-20:
> @chirag-silabs Is this the same as: #20862 ?

No that is related to the socket based implementation while the setup here works on the LwIP implementation. The LwIP implementation was fixed #20136 .

chirag-silabs, 2022-07-20:
Okay I have followed these steps. Still getting the same thing. 

./chip-tool accesscontrol write acl '[{"fabricIndex": 1, "privilege": 5, "authMode": 2, "subjects": [112233], "targets": null },{"fabricIndex": 1, "privilege": 4, "authMode": 3, "subjects": [1], "targets": null }]' 1 0

./chip-tool groupkeymanagement key-set-write '{"groupKeySetID": 42, "groupKeySecurityPolicy": 0, "epochKey0": "d0d1d2d3d4d5d6d7d8d9dadbdcdddedf", "epochStartTime0": 2220000,"epochKey1": "d1d1d2d3d4d5d6d7d8d9dadbdcdddedf", "epochStartTime1": 2220001,"epochKey2": "d2d1d2d3d4d5d6d7d8d9dadbdcdddedf", "epochStartTime2": 2220002 }' 1 0

./chip-tool groups add-group 0x0001 grp1 1 1

./chip-tool groupkeymanagement write group-key-map '[{"groupId": 1, "groupKeySetID": 42, "fabricIndex": 1}]' 1 0

./chip-tool groupsettings add-group grp1 0x0001

./chip-tool groupsettings add-keysets 0x0042 0 0x000000000021dfe0 hex:d0d1d2d3d4d5d6d7d8d9dadbdcdddedf

./chip-tool groupsettings bind-keyset 0x0001 0x0042

./chip-tool onoff toggle 0xffffffffffff0001 1

I see that the add-group command shows a non zero status. Please find the logs below. 
[groups.txt](https://github.com/project-chip/connectedhomeip/files/9150568/groups.txt)
[groups_dut.txt](https://github.com/project-chip/connectedhomeip/files/9150569/groups_dut.txt)

closed 2022-07-20 by chirag-silabs