project-chip/connectedhomeip #20991: [TC-DRLK-2.9] Door-lock Get Credential Command is returning INVALID_COMMAND .

Ashwinigrl, 2022-07-20:
**1.Problem**- In TC-DRLK-2.9, In Get Credential Command the value for credentialIndex in the range of  11 to 65534 is returning INVALID_COMMAND instead of NULL values.
 
 When the credentialIndex value is in the range of 1 to 10 is returning the NULL value even it is not set but if the value is between 11 to 65534 it is returning as INVALID_COMMAND instead of NULL value.

 **2. steps to reproduce**
 
> ./chip-tool doorlock clear-credential  '{ "credentialType" : 1 , "credentialIndex" : 65534 }'  1 1 --timedInteractionTimeoutMs 1000
> 
> ./chip-tool doorlock get-credential-status '{ "credentialType" : 1 , "credentialIndex" : 65534 }'  1 1 --timedInteractionTimeoutMs 1000
> 
> [1654698175.244136][5478:5483] CHIP:DMG:                                        status = 0x85 (INVALID_COMMAND),

**3.Test Environment**
App Used - lock app
Platform - Chip-tool - RPI-4, 8GB RAM
DUT - RPI-4, 8GB RAM
Network - Ethernet
commit id - 742b80e93346926435743976271ebf1362a3a85c

**4.logs**
[get credential cmd logs.txt](https://github.com/project-chip/connectedhomeip/files/9149010/get.credential.cmd.logs.txt)

bzbarsky-apple, 2022-07-20:
@Morozov-5F

woody-apple, 2022-07-20:
Cert Blocker Review: Assigning to @Morozov-5F for now, pending a new owner.

mykrupp, 2022-07-25:
I will take on this one.

woody-apple, 2022-07-29:
Cert Blocker Review: Marking this as 'needs provisional' unless we have a resolution to this issue today.

closed 2022-08-03 by woody-apple