project-chip/connectedhomeip #21007: ValidateCertificateChain is not spec complaint

msandstedt, 2022-07-20:
#### Problem

The spec description of `Crypto_VerifyChainDER` has this special provision, which is not standard x509 path validation:

> The Validity period validation for the root and optional intermediate certificates is performed
against the notBefore timestamp of the end-entity (leaf certificate) used as value for the current
time.

The sdk does actually implement these checks:

https://github.com/project-chip/connectedhomeip/blob/6d693f78b598426e2782a4ba3a5087e62d676e34/src/credentials/attestation_verifier/DefaultDeviceAttestationVerifier.cpp#L233

However, subsequently, `ValidateCertificateChain` also executes the standard x509 validity period check, which is different than the above.  The RFC-5280 x509 check is:

> To meet this goal, the path validation process verifies, among other things, that a prospective certification path (a sequence of n certificates) satisfies the following conditions:
> ...
>      (d)  for all x in {1, ..., n}, the certificate was valid at the
>         time in question.

This occurs because the OpenSSL and mbedTLS back ends do this automatically.  The OpenSSL check is described here: https://www.openssl.org/docs/man3.0/man1/openssl-verification-options.html.  The mbedTLS check occurs here: https://github.com/Mbed-TLS/mbedtls/blob/ecc1031dbfcaef88f19dd8cafb7451b3b7617a1b/library/x509_crt.c#L2884

 #### Proposed Solution

A verifier callback can be injected into OpenSSL to modify the expiration policy for spec compliance.  mbedTLS does not appear to have such a ready mechanism and might require patching or forking.

turon, 2022-07-20:
Isn't this a spec issue?  "not standard x509 path validation" sounds scary or wrong.

emargolis, 2022-08-10:
Closing with #21531

closed 2022-08-10 by emargolis