project-chip/connectedhomeip #21009: [TC-SC-3.6] - Reference app required to make DUT as controller

somu1710, 2022-07-20:
**Description:**

- In Secure channel Test case[TC-SC-3.6], requires to verify DUT’s CaseSessionsPerFabric in CapabilityMinima is at least 3 in 5 different fabrics in which each fabric contains 3 reference devices.

- To execute this test case as it is, requires `15 TH's `and `one DUT. `

- If we develop the reference app, which can act as a controller, these controllers can be run  (bring-up) on the same device to commission the DUT.
 
- In a RPI, Connectedhomeip folder, we have Chip-tool, All-clusters-app, lock-app, light-app etc, if these apps can act as a controller, this can send a  subscribe message from the same device to the DUT. 

- Our Raspberrypi device [All-cluster-app] does not  have an provision to handle this test case.Hence we request for an reference app.

**Test case ref:** 
https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/securechannel.adoc#tc-sc-3-6-case-resource-validation

woody-apple, 2022-07-22:
Cert Blocker Review: @cjandhyala to follow up with @vivien-apple for possibly how to implement this using the tools we have today.

tcarmelveilleux, 2022-07-27:
We already stated elsewhere that a bespoke manner to use a "DUT as a controller" does not go in the spirit of what is required, and actually what we need is just to have N chip-tool clients concurrently, or N python clients concurrently. 

There is no arbitrary control path that we could build in less time than using proper means either with chip-tool or Python.

@mrjerryjohns for color about multiple controllers per instance

woody-apple, 2022-07-29:
Cert Blocker Review: Marking this as 'needs provisional' unless we have a resolution to this issue today.

cjandhyala, 2022-08-04:
Hi @tcarmelveilleux @mrjerryjohns , Could you share the progress on getting the python script to test TC-SC-3.6? The same solution could be used for TC-RR-1.1 and a test step in TC-IDM-4.2 (which also requires having a controller on the same fabric).

tcarmelveilleux, 2022-08-04:
I am making extremely good progress on TC-RR-1.1.

I will attempt to start TC-SC-3.6 today.

woody-apple, 2022-08-16:
SDK Review: We are not taking SDK changes at this point for this test case. Moving to post 1.0.

closed 2022-08-16 by woody-apple