project-chip/connectedhomeip #21017: [TC-IDM-4.3] KeepSubscription not working

pankore, 2022-07-20:
#### Problem
- TC-IDM-4.3 step 15
- TH sends a subscription request action for an attribute to the DUT with the KeepSubscriptions flag set to True. Activate the subscription between DUT and the TH. Initiate another subscription request action to the DUT for another attribute with the KeepSubscriptions flag set to False. Change both the attribute values on the DUT.
- DUT did not receive notification for the first subscription
- See log below

[tc-idm-4.3-step15.txt](https://github.com/project-chip/connectedhomeip/files/9153799/tc-idm-4.3-step15.txt)

bzbarsky-apple, 2022-07-21:
@pankore If it has those steps in that order, then yes.

franck-apple, 2022-07-21:
@pankore looks like a test plan issue, so closing the SDK issue. Please file an issue against the test case repo.

closed 2022-07-21 by franck-apple