project-chip/connectedhomeip #21018: [TC-TSTAT-2.2] Writing certain values within occupied-cooling-setpoint range fail

pankore, 2022-07-20:
#### Problem
- `./chip-tool thermostat read min-cool-setpoint-limit 1 1` returns 1600
- `./chip-tool thermostat write occupied-cooling-setpoint 1600 1 1` fails
- Writing values 1600 up to 2400 fail, writing 2400 will pass

[tstat-occupied-cooling-setpoint.txt](https://github.com/project-chip/connectedhomeip/files/9153973/tstat-occupied-cooling-setpoint.txt)

bzbarsky-apple, 2022-07-21:
@pankore Does the server have the AUTO feature enabled?  If so, there are lower-bound constraints on OccupiedCoolingSetpoint other than MinCoolSetpointLimit.

pankore, 2022-07-21:
Thanks, I overlooked this in the specs. My auto feature is enabled. Closing this issue.

closed 2022-07-21 by pankore