project-chip/connectedhomeip #21069: [SVE1][TC-IDM-4.1] darwin-framework-tool Support for `--data-version` argument for `on-off subscribe` command

liamgonyea, 2022-07-21:
#### Problem
`darwin-framework-tool` support is needed for TC-IDM-4.1 Step 11 to "start another subscription with the DataVersionFilter field set". Current implementation does not show support for this argument in `darwin-framework-tool`
```
>>> onoff  subscribe on-off   100 1000  1 1 --data-version 0x869d6837 
[1658428507877] [7406:4378478] CHIP: [TOO] InitArgs: Invalid argument endpoint-id: --data-version
Usage:
   onoff subscribe attr-name min-interval max-interval wait node-id endpoint-id [--commissioner-name] [--fabric-filtered] [--keepSubscriptions]
[1658428507877] [7406:4378478] CHIP: [TOO] Run command failure: ../../third_party/connectedhomeip/examples/chip-tool/commands/common/Commands.cpp:144: Error 0x0000002F
[1658428508044] [7406:4400075] CHIP: [DMG] Subscription Liveness timeout with SubscriptionID = 0xf111855e, Peer = 01:0000000000000001

```
 
[image]

[[TC-IDM-4.1] SubscriptionRequestMessage from DUT test cases. [{DUT_Client}]
](https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/interactiondatamodel.adoc#341-tc-idm-41-subscriptionrequestmessage-from-dut-test-cases-dut_client)
#### Proposed Solution
Add support similar to `chip-tool`:
```
Usage:
   onoff subscribe attr-name min-interval max-interval destination-id endpoint-ids [--paa-trust-store-path] [--commissioner-name] [--commissioner-nodeid] [--trace_file] [--trace_log] [--trace_decode] [--ble-adapter] [--fabric-filtered] [--data-version] [--keepSubscriptions] [--timeout]

...

[--data-version]:
  Comma-separated list of data versions for the clusters being subscribed to.

```

#### Testing setup
commit `69bfcc7de9f4c6f616ba8d70beb78d9b93cc3c25` (HEAD -> sve_branch, origin/sve)

bzbarsky-apple, 2022-07-21:
The Darwin framework MUST NOT add support for sending a data version with read/subscribe requests until it adds support for reporting data versions along with read/subscribed data.  Which right now it does not have.  And it's not required to have any of that support, afaict....

Filed https://github.com/CHIP-Specifications/chip-test-plans/issues/1894 on the test plan.

liamgonyea, 2022-08-11:
Makes sense to me, https://github.com/CHIP-Specifications/chip-test-plans/issues/1894 addresses the issue with PICS, closing issue

closed 2022-08-11 by liamgonyea