project-chip/connectedhomeip #21089: [SVE1][TC-CNET-4.20] TH All-clusters-app unable to scan for surrounding networks

liamgonyea, 2022-07-22:
#### Problem
When testing DM (Device Management) cluster TC-CNET-4.20, Step 1, chip-tool runs `./chip-tool networkcommissioning scan-networks 1 0` to have TH=all-clusters-app scan for available networks. It cannot find any available networks so then Steps 3, 4, and 6 fail because there are no networks.

Currently at SVE1 building with at least 2 known routers in addition to the one the raspi/mac are connected to (ssid: TC)
[image]

#### Log snippet

Received Command Response Data, Endpoint=0 Cluster=0x0000_0031 Command=0x0000_0001
[1658448070313] [11485:4699971] CHIP: [TOO]     networkingStatus: 12
[1658448070313] [11485:4699971] CHIP: [TOO]     wiFiScanResults: 0 entries
[1658448070313] [11485:4699971] CHIP: [TOO]    }

 #### Test Setup
**chip-tool:** 
running on mac, commit `69bfcc7de9f4c6f616ba8d70beb78d9b93cc3c25` (HEAD -> sve_branch, origin/sve)

**all-clusters-app:** `./out/ble/all-clusters-app/chip-all-clusters-app --wifi `
commit 69bfcc7de9f4c6f616ba8d70beb78d9b93cc3c25 (HEAD -> sve_branch, origin/sve)
running on a RasPi connected over wifi to a router that has the mac running chip-tool connected to it. RasPi is not connected with an ethernet cable.

#### Logs
[TC-CNET-4.20 DUT Logs .txt](https://github.com/project-chip/connectedhomeip/files/9163513/TC-CNET-4.20.DUT.Logs.txt)
[TC-CNET-4.20 TH Logs .txt](https://github.com/project-chip/connectedhomeip/files/9163514/TC-CNET-4.20.TH.Logs.txt)

bzbarsky-apple, 2022-07-22:
The ScanNetworksResponse is coming back with `UnknownError`, right?

Does the RasPi have wpa_supplicant installed @liamgonyea ?

bzbarsky-apple, 2022-07-22:
Fundamentally, `ConnectivityManagerImpl::StartWiFiScan` just doesn't have nearly enough logging on the error paths...

closed 2023-02-04 by stale[bot]