project-chip/connectedhomeip #21096: [SVE1][TC-ACL-2.10] Step 8 Return CONSTRAINT_ERROR rather than SUCCESS

AlvinHsiao, 2022-07-22:
## Problem
Step 8 Return CONSTRAINT_ERROR rather than SUCCESS.

## Steps:
./chip-tool accesscontrol write extension '[{"data":"1000110010001100"}]' 11 0

## system configuration :
Chip-Tool: RPI
DUT used : Infineon CYW30739
app use: Lighting-app
chip-tool and DUT commit id : https://github.com/project-chip/connectedhomeip/commit/d24168c36f505cb0b710095452b1422b477c018c (sve SHA)

## Logs:
[TC-ACL-2.10_TH.txt](https://github.com/project-chip/connectedhomeip/files/9165225/TC-ACL-2.10_TH.txt)
[TC-ACL-2.10_DUT.txt](https://github.com/project-chip/connectedhomeip/files/9165227/TC-ACL-2.10_DUT.txt)

bzbarsky-apple, 2022-07-22:
Spec says:

> This field MAY be used by manufacturers to store arbitrary TLV-encoded data related to a fabric’s Access Control Entries.

> The contents SHALL consist of a top-level anonymous list; each list element SHALL include a profile-specific tag encoded in fully-qualified form.

The data being passed in here is not a TLV list, so the server returns CONSTRAINT_ERROR.  Where did this `"1000110010001100"` data come from @AlvinHsiao ?

closed 2022-07-22 by bzbarsky-apple

AlvinHsiao, 2022-07-25:
@bzbarsky-apple It came from SVE1_TestPlanVerificationSteps_V1.3.

bzbarsky-apple, 2022-07-26:
Filed https://github.com/CHIP-Specifications/chip-test-plans/issues/1930