project-chip/connectedhomeip #21099: [TC-SU-2.2] When we try to update with older version or same version, transfer of image should never happen.

manogna-silabs, 2022-07-22:
Description: When we try to update with older version or same version, transfer of image should never happen.

Steps to reproduce the issue:
Provider:
1.  ./src/app/ota_image_tool.py create -v 0xFFF1 -p 0x8005 -vn 2 -vs "1.0" -da sha256 ota.gbl otaImage.ota

Chip-tool:
2. ./chip-tool pairing onnetwork 1 20202021

STEP-4:
./chip-tool pairing onnetwork-long 321 20202021 3841

STEP-5:

./chip-tool accesscontrol write acl '[{"fabricIndex": 1, "privilege": 5, "authMode": 2, "subjects": [112233], "targets": null}, {"fabricIndex": 1, "privilege": 3, "authMode": 2, "subjects": null, "targets": null}]' 1 0

STEP 6: 
Perform commissioning
./chip-tool  pairing ble-wifi 2 SSID password 20202021 3840

STEP-6:
./chip-tool otasoftwareupdaterequestor announce-ota-provider 1 0 0 0 2 0 

 Verify there is a transfer of the software image from the TH/OTA-P to the DUT

 STEP 7:
After 1st transfer,  resend the announce command

Expected outcome:
Transfer should never happen for the same version/older version.

Actual:
Transfer is happening.

Logs:
[AnnounceLogs.txt](https://github.com/project-chip/connectedhomeip/files/9165487/AnnounceLogs.txt)
[DUTLogs.txt](https://github.com/project-chip/connectedhomeip/files/9165488/DUTLogs.txt)
[providerLogs.txt](https://github.com/project-chip/connectedhomeip/files/9165489/providerLogs.txt)

Note: This issue is tested on previous SVE branch and on latest SVE branch with the
 Commit ID: ea7a692120965623ee52c9805ffa2008b2af6b3d. 

Thanks,
G. Manogna

bzbarsky-apple, 2022-07-22:
@tcarmelveilleux @carol-apple @selissia

woody-apple, 2022-07-22:
Cert Blocker Review: @selissia to take a look

selissia, 2022-07-22:
It doesn't look like the update image was built with CHIP_DEVICE_CONFIG_DEVICE_SOFTWARE_VERSION set to 2. @manogna-silabs , please double-check

manogna-silabs, 2022-07-25:
Set CHIP_DEVICE_CONFIG_DEVICE_SOFTWARE_VERSION to some version say 2 and build the .ota file with older version/same version. 
 DUT does not start transferring the software image. It is as expected. 
 Hence closing the issue.
 
 Thanks,
 G. Manogna.

closed 2022-07-25 by manogna-silabs