project-chip/connectedhomeip #21107: [EFR32] Door lock feature map is return wrong value

shgutte, 2022-07-22:
#### Problem
Feature map suppose to return the future map value as 435 According to the test plan and Test harness 

Actual output:
FeatureMap: 3507

Expected output:
FeatureMap: 435

Tested on sve branch commit ID: bcdfbfa7f3fea45e32f2e5acb4f793bb4ac8a66b

App used - lock app
Platform - Chip-tool - RPI-4, 8GB RAM
DUT - EFR32 + WF200 , EFR32 +RS9116
Network - Ble-wifi

Chip-tool-logs 
[feature-map-error.txt](https://github.com/project-chip/connectedhomeip/files/9168059/feature-map-error.txt)

bzbarsky-apple, 2022-07-22:
`435 == 0b000110110011` means PIN, RID, SCH, DPS, COTA, USR.

`3507 == 0b110110110011` means PIN, RID, WDSCH, DPS, COTA, USR, YDSCH, HDSCH after https://github.com/CHIP-Specifications/connectedhomeip-spec/pull/5312 is applied.  the SDK implements the state after that PR.

woody-apple, 2022-07-22:
Cert Blocker Review: The SDK implements the coming version of the spec in the PR @bzbarsky-apple lists above. We believe no further SDK changes are needed, please re-test when the spec is updated.

closed 2022-07-22 by woody-apple