project-chip/connectedhomeip #21125: Client-side logic doesn't send `InvalidSubscription` error back correctly

mrjerryjohns, 2022-07-22:
#### Problem

If a report is received that doesn't match any existing subs present on the client, the SDK today doesn't send back an `InvalidSubscription` error. Instead, it just sends back a StatusResponse containing success.

This violates the spec.

yunhanw-google, 2022-07-25:
https://github.com/project-chip/connectedhomeip/blob/master/src/app/ReadClient.cpp#L471,  it seems we are sending status failure if that is invalid subscription in client side, it seems that we just need to send InvalidSubscription status.

yunhanw-google, 2022-08-03:
fixed in 
https://github.com/project-chip/connectedhomeip/pull/21368
https://github.com/project-chip/connectedhomeip/pull/21500

closed 2022-08-03 by yunhanw-google