project-chip/connectedhomeip #21129: [SVE1][TC-CNET-4.20] TH All-clusters-app crashes when receiving wifi connect request

cm-s-dev, 2022-07-22:
**Problem**
For CNET 4.20, step 5 send connect-network command causes TH to crash with segfault after receiving wifi connect information. 

**TH Log Snippet**
`CHIP:DMG: Received command for Endpoint=0 Cluster=0x0000_0031 Command=0x0000_0006
CHIP:NP: LinuxNetworkCommissioningDelegate: SSID: MyTest$▒֠▒,;▒Q▒▒}▒▒u▒1824ff0118
./allclust.sh: line 1:  1431 Segmentation fault      sudo ./apps/all-cluster-app --trace_file /tmp/trace_file.log --trace_decode 1 --discriminator 2095 --wifi`

**Test setup:**
RPI TH CSA image v1.3 `aa9457e6b94b735076dff6297176183bf9780177` and DUT

**TH log:**
[TH1.3_rpi_cnet-4.20_sve1_crash.txt](https://github.com/project-chip/connectedhomeip/files/9170421/TH1.3_rpi_cnet-4.20_sve1_crash.txt)

cjandhyala, 2022-07-29:
@cm-s-dev Could you provide the stack trace log using gdb ?

woody-apple, 2022-07-29:
Cert Blocker Review: Closing this, as the CSG is not seeing this fail in the reference platform.

closed 2022-07-29 by woody-apple