project-chip/connectedhomeip #21130: [crash] Matter:chip::Controller::DeviceCommissioner::CommissioningStageComplete

aajain-com, 2022-07-22:
#### Problem
Crash seen during stability testing with SHA `aa9457e6b94b735076dff6297176183bf9780177`

```
Exception Type:  EXC_BAD_ACCESS (SIGSEGV)
Exception Subtype: KERN_INVALID_ADDRESS at 0x002000010704c018 -> 0x000000010704c018 (possible pointer authentication failure)
Exception Codes: 0x0000000000000001, 0x002000010704c018
VM Region Info: 0x10704c018 is in 0x107000000-0x107800000;  bytes after start: 311320  bytes before end: 8077287
      REGION TYPE                 START - END      [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      __LINKEDIT               106d20000-106fe0000 [ 2816K] r--/rw- SM=COW  ...ework/HomeKit
      GAP OF 0x20000 BYTES
--->  MALLOC_SMALL             107000000-107800000 [ 8192K] rw-/rwx SM=PRV  
      mapped file              107800000-109500000 [ 29.0M] r--/r-- SM=ALI  ...t_id=9f847945
Termination Reason: SIGNAL 11 Segmentation fault: 11
```

```
0   Matter                        	       0x10425199c chip::Controller::DeviceCommissioner::CommissioningStageComplete(chip::ChipError, chip::Controller::CommissioningDelegate::CommissioningReport) + 108 (connectedhomeip/src/controller/CHIPDeviceController.cpp:1557)
1   Matter                        	       0x104255160 chip::Controller::OnBasicFailure(void*, chip::ChipError) + 168 (connectedhomeip/src/controller/CHIPDeviceController.cpp:1458)
2   Matter                        	       0x104255160 chip::Controller::OnBasicFailure(void*, chip::ChipError) + 168 (connectedhomeip/src/controller/CHIPDeviceController.cpp:1458)
3   Matter                        	       0x104075c74 std::__1::__function::__func<chip::ChipError chip::Controller::ClusterBase::InvokeCommand<chip::app::Clusters::GeneralCommissioning::Commands::ArmFailSafe::Type>(chip::app::Clusters::GeneralCommissioning::Commands::ArmFailSafe::Type const&, void*, void (*)(void*, chip::app::Clusters::GeneralCommissioning::Commands::ArmFailSafe::Type::ResponseType const&), void (*)(void*, chip::ChipError), chip::Optional<unsigned short> const&)::'lambda'(chip::ChipError), std::__1::allocator<chip::ChipError chip::Controller::ClusterBase::InvokeCommand<chip::app::Clusters::GeneralCommissioning::Commands::ArmFailSafe::Type>(chip::app::Clusters::GeneralCommissioning::Commands::ArmFailSafe::Type const&, void*, void (*)(void*, chip::app::Clusters::GeneralCommissioning::Commands::ArmFailSafe::Type::ResponseType const&), void (*)(void*, chip::ChipError), chip::Optional<unsigned short> const&)::'lambda'(chip::ChipError)>, void (chip::ChipError)>::operator()(chip::ChipError&&) + 44 connectedhomeip/src/controller/CHIPCluster.h:83)
4   Matter                        	       0x10407588c chip::Controller::TypedCommandCallback<chip::app::Clusters::GeneralCommissioning::Commands::ArmFailSafeResponse::DecodableType>::OnError(chip::app::CommandSender const*, chip::ChipError) + 96
5   Matter                        	       0x10420cc30 chip::app::CommandSender::OnResponseTimeout(chip::Messaging::ExchangeContext*) + 232 (connectedhomeip/src/app/CommandSender.cpp:214)
6   Matter                        	       0x10430d3e4 chip::Messaging::ExchangeContext::NotifyResponseTimeout(bool) + 128 (connectedhomeip/src/messaging/ExchangeContext.cpp:457)
7   Matter                        	       0x10430d484 chip::Messaging::ExchangeContext::HandleResponseTimeout(chip::System::Layer*, void*) + 68 (connectedhomeip/src/messaging/ExchangeContext.cpp:445)
8   Matter                        	       0x10432a278 chip::System::TimerData::Callback::Invoke() const + 40 (connectedhomeip/src/system/SystemTimer.h:61)
9   Matter                        	       0x104329aa8 chip::System::TimerPool<chip::System::TimerList::Node>::Invoke(chip::System::TimerList::Node*) + 88 (connectedhomeip/src/system/SystemTimer.h:224)
10  Matter                        	       0x104328698 chip::System::LayerImplSelect::HandleTimerComplete(chip::System::TimerList::Node*) + 60 (connectedhomeip/src/system/SystemLayerImplSelect.cpp:419)
11  Matter                        	       0x104328650 ___ZN4chip6System15LayerImplSelect10StartTimerENSt3__16chrono8durationIjNS2_5ratioILl1ELl1000EEEEEPFvPNS0_5LayerEPvESA__block_invoke + 80 (connectedhomeip/src/system/SystemLayerImplSelect.cpp:147)
```

The only two relevant log lines I could find in the the system at the time of crash.
```
2022-07-19 14:41:50.0272 -0700 default 0x08e5d 0x0 homed[172] (Matter): 🔵 [1658266910040] [172:36445] CHIP: [DMG] Time out! failed to receive invoke command response from Exchange: 33122i
2022-07-19 14:41:50.0272 -0700 default 0x08e5d 0x0 homed[172] (Matter): 🔵 [1658266910040] [172:36445] CHIP: [CTL] Received failure response connectedhomeip/src/app/CommandSender.cpp:214: CHIP E
```

Apple reference: rdar://97444152

mrjerryjohns, 2022-07-25:
@cecille this looks pretty gnarly, and I have a sneaky suspicion this may not be Darwin specific. Perhaps you could try spoofing an exchange timeout to re-create this situation when using the auto-commissioner and see if things break?

cecille, 2022-07-25:
Would you be able to share more of the logs? I'm not sure what stage this failed on and haven't been able to repro this here (tested timing out a couple of stages, but it would help to narrow down).

Also, are there changes to SDK in this test? I checked out that SHA, but the listed line numbers don't match up.

aajain-com, 2022-07-27:
Unfortunately the system that collects the logs only stores the snapshots of the logs at that time so I am unable to find more logs.

woody-apple, 2022-08-30:
This is no longer showing up in crashes.

closed 2022-08-30 by woody-apple