project-chip/connectedhomeip #21131: DeviceInfoProvider leaks storage

tcarmelveilleux, 2022-07-22:
#### Problem
- DeviceInfoProvider allows the UserLabel list to be set per endpoint
  - Default impl in `examples/providers/DeviceInfoProviderImpl.cpp`  stores current list size as one key, and all the entries for that endpoint as separate keys
 
The problems this causes are as follows:
- If the list is written with N entries, it takes N+1 PersistentStorageDelegate entries instead of possibly 1-2: 1 for the length, 1 for the list of entries, or better, 1 for the whole set of entries
- If the list is written with 0 entries, the size changes, but impl in `examples/providers/DeviceInfoProviderImpl.cpp` does not delete or move any of the storage: all the storage is left behind.
- If the RemoveFabric is done on the last fabric, all the data is left behind, which violates the following paragraph of spec: 

> 11.18.7.12. RemoveFabric Command
> ...
> - If the FabricIndex matches the last remaining entry in the Fabrics list, then the device SHALL delete all Matter related data on the node which was created since it was commissioned. This includes all Fabric-Scoped data, including Access Control List, bindings, scenes, group keys, operational certificates, etc.
> 

There is no way to delete any of the data on any of the endpoints. As soon as a userlabel is written, the key is persisted until the platform removes all storage.

 #### Proposed Solution
- Add APIs to the interface to handle removal of last fabric
- When the list becomes smaller than previous list for a given endpoint, all "over-size" keys (keys for [current_length..previous_length-1]) should be deleted, to recover space.
- Consider a design that uses 1 key per endpoint, rather than 1 key per label.

mrjerryjohns, 2022-07-25:
@yufengwangca could you help take a look at this please?

yufengwangca, 2022-07-28:
#21336 addressed the following two proposals which address the existing problems
```
- Add APIs to the interface to handle removal of last fabric
- When the list becomes smaller than previous list for a given endpoint, all "over-size" keys (keys for [current_length..previous_length-1]) should be deleted, to recover space.
```

For `- Consider a design that uses 1 key per endpoint, rather than 1 key per label.`

I was wondering if we need to do this refactor, since we need to introduce extra data structure and meta date to manage the label list, and we can not add the item directly when receive ListOperation::AppendItem command, we need to read the previous list and append at the last and write back the whole list to storage.

closed 2022-07-29 by yufengwangca