project-chip/connectedhomeip #21134: SVE1 testing: TC-MediaInput-3.17 rename-input output shows 0x01 failure response

Garik098, 2022-07-22:
As per the test step 1 in TC-MediaInput-3.17, ./chip-tool mediainput rename-input 1 'DVD' 1 1 command output is giving failure response while the expected outcome should be a success response. 

Below is the log for the same. 

shyam@shyams-MacBook-Pro debug % ./chip-tool mediainput rename-input 1 'DVD' 1 1
[1658524832134] [23667:587168] CHIP: [DL] _Init
[1658524832134] [23667:587168] CHIP: [IN] UDP::Init bind&listen port=0
[1658524832135] [23667:587168] CHIP: [IN] UDP::Init bound to port=57812
[1658524832135] [23667:587168] CHIP: [IN] BLEBase::Init - setting/overriding transport
[1658524832135] [23667:587168] CHIP: [IN] TransportMgr initialized
[1658524832135] [23667:587168] CHIP: [FP] Initializing FabricTable from persistent storage
[1658524832135] [23667:587168] CHIP: [TS] Last Known Good Time: 2022-07-11T10:23:09
[1658524832135] [23667:587168] CHIP: [FP] Fabric index 0x1 was retrieved from storage. Compressed FabricId 0xE2E58F6F9CD07DB0, FabricId 0x0000000000000001, NodeId 0x000000000001B669, VendorId 0xFFF1
[1658524832136] [23667:587168] CHIP: [FP] Fabric index 0x2 was retrieved from storage. Compressed FabricId 0x328FFFEB8E8ABEBB, FabricId 0x0000000000000002, NodeId 0x000000000001B669, VendorId 0xFFF1
[1658524832136] [23667:587168] CHIP: [FP] Fabric index 0x3 was retrieved from storage. Compressed FabricId 0x2CB5E5948E4A6EE8, FabricId 0x0000000000000003, NodeId 0x000000000001B669, VendorId 0xFFF1
[1658524832138] [23667:587168] CHIP: [ZCL] Using ZAP configuration...
[1658524832138] [23667:587168] CHIP: [CTL] System State Initialized...
[1658524832138] [23667:587168] CHIP: [CTL] Stopping commissioning discovery over DNS-SD
[1658524832138] [23667:587168] CHIP: [CTL] Setting attestation nonce to random value
[1658524832138] [23667:587168] CHIP: [CTL] Setting CSR nonce to random value
[1658524832138] [23667:587168] CHIP: [CTL] Stopping commissioning discovery over DNS-SD
[1658524832138] [23667:587168] CHIP: [CTL] Setting attestation nonce to random value
[1658524832138] [23667:587168] CHIP: [CTL] Setting CSR nonce to random value
[1658524832140] [23667:587168] CHIP: [CTL] Generating NOC
[1658524832143] [23667:587168] CHIP: [FP] Validating NOC chain
[1658524832154] [23667:587168] CHIP: [FP] NOC chain validation successful
[1658524832154] [23667:587168] CHIP: [FP] Updated fabric at index: 0x1, Node ID: 0x000000000001B669
[1658524832154] [23667:587168] CHIP: [TS] Last Known Good Time: 2022-07-11T10:23:09
[1658524832154] [23667:587168] CHIP: [TS] New proposed Last Known Good Time: 2021-01-01T00:00:00
[1658524832154] [23667:587168] CHIP: [TS] Retaining current Last Known Good Time
[1658524832157] [23667:587168] CHIP: [FP] Metadata for Fabric 0x1 persisted to storage.
[1658524832159] [23667:587168] CHIP: [TS] Committing Last Known Good Time to storage: 2022-07-11T10:23:09
[1658524832160] [23667:587168] CHIP: [CTL] Joined the fabric at index 1. Compressed fabric ID is: 0x0000000000000000
[1658524832160] [23667:587168] CHIP: [CTL] Stopping commissioning discovery over DNS-SD
[1658524832160] [23667:587168] CHIP: [CTL] Setting attestation nonce to random value
[1658524832160] [23667:587168] CHIP: [CTL] Setting CSR nonce to random value
[1658524832163] [23667:587168] CHIP: [CTL] Generating NOC
[1658524832166] [23667:587168] CHIP: [FP] Validating NOC chain
[1658524832178] [23667:587168] CHIP: [FP] NOC chain validation successful
[1658524832178] [23667:587168] CHIP: [FP] Updated fabric at index: 0x2, Node ID: 0x000000000001B669
[1658524832178] [23667:587168] CHIP: [TS] Last Known Good Time: 2022-07-11T10:23:09
[1658524832178] [23667:587168] CHIP: [TS] New proposed Last Known Good Time: 2021-01-01T00:00:00
[1658524832178] [23667:587168] CHIP: [TS] Retaining current Last Known Good Time
[1658524832180] [23667:587168] CHIP: [FP] Metadata for Fabric 0x2 persisted to storage.
[1658524832181] [23667:587168] CHIP: [TS] Committing Last Known Good Time to storage: 2022-07-11T10:23:09
[1658524832182] [23667:587168] CHIP: [CTL] Joined the fabric at index 2. Compressed fabric ID is: 0x0000000000000000
[1658524832182] [23667:587168] CHIP: [CTL] Stopping commissioning discovery over DNS-SD
[1658524832182] [23667:587168] CHIP: [CTL] Setting attestation nonce to random value
[1658524832182] [23667:587168] CHIP: [CTL] Setting CSR nonce to random value
[1658524832185] [23667:587168] CHIP: [CTL] Generating NOC
[1658524832189] [23667:587168] CHIP: [FP] Validating NOC chain
[1658524832200] [23667:587168] CHIP: [FP] NOC chain validation successful
[1658524832200] [23667:587168] CHIP: [FP] Updated fabric at index: 0x3, Node ID: 0x000000000001B669
[1658524832200] [23667:587168] CHIP: [TS] Last Known Good Time: 2022-07-11T10:23:09
[1658524832200] [23667:587168] CHIP: [TS] New proposed Last Known Good Time: 2021-01-01T00:00:00
[1658524832200] [23667:587168] CHIP: [TS] Retaining current Last Known Good Time
[1658524832202] [23667:587168] CHIP: [FP] Metadata for Fabric 0x3 persisted to storage.
[1658524832203] [23667:587168] CHIP: [TS] Committing Last Known Good Time to storage: 2022-07-11T10:23:09
[1658524832204] [23667:587168] CHIP: [CTL] Joined the fabric at index 3. Compressed fabric ID is: 0x0000000000000000
[1658524832224] [23667:587171] CHIP: [DL] _OnPlatformEvent
[1658524832224] [23667:587171] CHIP: [TOO] Sending command to node 0x1
[1658524832224] [23667:587171] CHIP: [CSM] FindOrEstablishSession: PeerId = E2E58F6F9CD07DB0:0000000000000001
[1658524832224] [23667:587171] CHIP: [CSM] FindOrEstablishSession: No existing OperationalDeviceProxy instance found
[1658524832224] [23667:587171] CHIP: [CTL] OperationalDeviceProxy[E2E58F6F9CD07DB0:0000000000000001]: State change 1 --> 2
[1658524832224] [23667:587171] CHIP: [DIS] Resolving E2E58F6F9CD07DB0:0000000000000001 ...
[1658524832224] [23667:587171] CHIP: [DL] Resolve type=_matter._tcp name=E2E58F6F9CD07DB0-0000000000000001 interface=0
[1658524832224] [23667:587171] CHIP: [DL] Mdns : OnNewInterface hostname:E45F012EEC9F0000.local. fullname:E2E58F6F9CD07DB0-0000000000000001._matter._tcp.local. interface: 7
[1658524832225] [23667:587171] CHIP: [DL] Mdns: OnNewAddress interface: 7 ip:fe80::e65f:1ff:fe2e:ec9f
[1658524832225] [23667:587171] CHIP: [DIS] Node ID resolved for E2E58F6F9CD07DB0:0000000000000001
[1658524832225] [23667:587171] CHIP: [DIS] 	Hostname: E45F012EEC9F0000
[... 101 lines cut ...]
[1658524832499] [23667:587168] CHIP: [CTL] Shutting down the commissioner
[1658524832499] [23667:587168] CHIP: [CTL] Shutting down the controller
[1658524832499] [23667:587168] CHIP: [IN] Expiring all sessions for fabric 0x2!!
[1658524832499] [23667:587168] CHIP: [FP] Forgetting fabric 0x2
[1658524832499] [23667:587168] CHIP: [TS] Pending Last Known Good Time: 2022-07-11T10:23:09
[1658524832499] [23667:587168] CHIP: [TS] Previous Last Known Good Time: 2022-07-11T10:23:09
[1658524832499] [23667:587168] CHIP: [TS] Reverted Last Known Good Time to previous value
[1658524832499] [23667:587168] CHIP: [CTL] Shutting down the commissioner
[1658524832499] [23667:587168] CHIP: [CTL] Shutting down the controller
[1658524832499] [23667:587168] CHIP: [IN] Expiring all sessions for fabric 0x3!!
[1658524832499] [23667:587168] CHIP: [FP] Forgetting fabric 0x3
[1658524832499] [23667:587168] CHIP: [TS] Pending Last Known Good Time: 2022-07-11T10:23:09
[1658524832499] [23667:587168] CHIP: [TS] Previous Last Known Good Time: 2022-07-11T10:23:09
[1658524832499] [23667:587168] CHIP: [TS] Reverted Last Known Good Time to previous value
[1658524832499] [23667:587168] CHIP: [CTL] Shutting down the System State, this will teardown the CHIP Stack
[1658524832499] [23667:587168] CHIP: [DMG] IM WH moving to [Uninitialized]
[1658524832499] [23667:587168] CHIP: [DMG] IM WH moving to [Uninitialized]
[1658524832499] [23667:587168] CHIP: [DMG] IM WH moving to [Uninitialized]
[1658524832499] [23667:587168] CHIP: [DMG] IM WH moving to [Uninitialized]
[1658524832499] [23667:587168] CHIP: [DMG] All ReadHandler-s are clean, clear GlobalDirtySet
[1658524832499] [23667:587168] CHIP: [BLE] CancelConnection
[1658524832499] [23667:587168] CHIP: [DL] Inet Layer shutdown
[1658524832499] [23667:587168] CHIP: [DL] BLE shutdown
[1658524832499] [23667:587168] CHIP: [DL] System Layer shutdown
[1658524832499] [23667:587168] CHIP: [TOO] Run command failure: IM Error 0x00000501: General error: 0x01 (FAILURE)

bzbarsky-apple, 2022-07-23:
@Garik098 What server app is this?

Garik098, 2022-08-02:
Working fine on the tv-app

closed 2022-08-02 by Garik098