project-chip/connectedhomeip #21139: `all-clusters-app` not correctly taking `--vendor-id` input into QR code generation

liamgonyea, 2022-07-22:
#### Problem
When running `./out/all-clusters-app/chip-all-clusters-app --vendor-id 6524` the `chip-all-clusters-app` does not change the value of the `--vendor-id` from the default `65521` to `6524 ` when displaying the QR/ Manual codes for pairing the sample accessory. Additionally, after the device is paired, the `--vendor-id` is still not reflected in `./darwin-framework-tool basic read vendor-id 1 0`

 #### Proposed Solution
Have all-clusters-app take the input given in the `--vendor-id` argument and use it in the app.

#### Logs
```
[1658520516.778673][2328:2328] CHIP:DL: NVS set: chip-factory/vendor-id = 6524 (0x197C)
[1658520516.778746][2328:2328] CHIP:DL: Device Configuration:
[1658520516.778827][2328:2328] CHIP:DL:   Serial Number: TEST_SN
[1658520516.778891][2328:2328] CHIP:DL:   Vendor Id: 65521 (0xFFF1)
[1658520516.778944][2328:2328] CHIP:DL:   Product Id: 32769 (0x8001)
[1658520516.779153][2328:2328] CHIP:DL:   Hardware Version: 0
[1658520516.779229][2328:2328] CHIP:DL:   Setup Pin Code (0 for UNKNOWN/ERROR): 20202021
[1658520516.779273][2328:2328] CHIP:DL:   Setup Discriminator (0xFFFF for UNKNOWN/ERROR): 3840 (0xF00)
[1658520516.779392][2328:2328] CHIP:DL:   Manufacturing Date: (not set)
[1658520516.779447][2328:2328] CHIP:DL:   Device Type: 65535 (0xFFFF)
[1658520516.779528][2328:2328] CHIP:-: ==== Onboarding payload for Standard Commissioning Flow ====
[1658520516.779598][2328:2328] CHIP:SVR: SetupQRCode: [MT:-24J0AFN00KA0648G00]
[1658520516.779651][2328:2328] CHIP:SVR: Copy/paste the below URL in a browser to see the QR Code:
[1658520516.779695][2328:2328] CHIP:SVR: https://project-chip.github.io/connectedhomeip/qrcode.html?data=MT%3A-24J0AFN00KA0648G00
[1658520516.779752][2328:2328] CHIP:SVR: Manual pairing code: [34970112332]
[1658520516.779798][2328:2328] CHIP:-: ==== Onboarding payload for Custom Commissioning Flows ====
[1658520516.779857][2328:2328] CHIP:SVR: SetupQRCode: [MT:-24J029Q00KA0648G00]
[1658520516.779910][2328:2328] CHIP:SVR: Copy/paste the below URL in a browser to see the QR Code:
[1658520516.779954][2328:2328] CHIP:SVR: https://project-chip.github.io/connectedhomeip/qrcode.html?data=MT%3A-24J029Q00KA0648G00
[1658520516.780011][2328:2328] CHIP:SVR: Manual pairing code: [749701123365521327694]
```
Using **SetupQRCode: [MT:-24J0AFN00KA0648G00]** from log above: 
```
$ ./chip-tool payload parse-setup-payload MT:-24J0AFN00KA0648G00
[1658533677613] [20597:5106998] CHIP: [SPL] Parsing base38Representation: MT:-24J0AFN00KA0648G00
[1658533677614] [20597:5106998] CHIP: [SPL] Version:       0
[1658533677614] [20597:5106998] CHIP: [SPL] VendorID:      65521
[1658533677614] [20597:5106998] CHIP: [SPL] ProductID:     32769
[1658533677614] [20597:5106998] CHIP: [SPL] Custom flow:   0    (STANDARD)
[1658533677614] [20597:5106998] CHIP: [SPL] Capabilities:  0x04 (On IP network)
[1658533677614] [20597:5106998] CHIP: [SPL] Discriminator: 3840
[1658533677614] [20597:5106998] CHIP: [SPL] Passcode:      20202021
```

bzbarsky-apple, 2022-07-23:
We store the vendor id in the configuration manager in `InitConfigurationManager`, but then we are using `GenericDeviceInstanceInfoProvider` which always returns `CHIP_DEVICE_CONFIG_DEVICE_VENDOR_ID`.

In `ChipLinuxAppInit` we should probably do something for the device instance info provider similar to what we do for the `CommissionableDataProvider`.  So stop storing these in the configuration manager altogether (remove `InitConfigurationManager`), implement a class that forwards most of `DeviceInstanceInfoProvider` to the default provider but overrides the vendord and product id to return the command-line values, if any.

andy31415, 2022-07-23:
why is this a cert blocker?

cjandhyala, 2022-07-25:
Commissioner tests (DUT as commissioner) need this in the reference app (all-cluster-app) to have the ability to launch the app with different vendor IDs and the commissioner is able to commission the device for values other than what is used as default in the reference apps.

bzbarsky-apple, 2022-07-27:
Fixing the `--vendor-id` and `--product-id` bits is not so bad.  But the device attestation certificate (which is selected at compile time right now) will then no longer match the claimed vendor/product id, and hence device attestation will fail...

closed 2022-07-27 by woody-apple