project-chip/connectedhomeip #21152: [TC-CADMIN-1.9] chip-tool will repeat to pair several times the DUT until timeout

wqx6, 2022-07-25:
#### Problem
In TC-CADMIN-1.9 we need to pairing the DUT on TH2 using the command `./chip-tool pairing code {node-id} {wrong-manual-code} --timeout 3`. The timeout will cause a problem.
If we set a short timeout, the chip-tool might return timeout before it resolve the address of the DUT and try to establish the PASESession.
But if we set a long timeout, the chip-tool will repeat to establish the PASESession several times with the DUT, and when the chip-tool fails to establish the PASESession 20 times, it can't pair the DUT anymore.

The test steps want us to send the command `./chip-tool pairing code {node-id} {wrong-manual-code} --timeout 3` 20 times instead of failing to establish the PASESession 20 times.

 #### Proposed Solution
Could we delete the timeout and count the times of establishing the PASESession until it reaches 20 and then use the right manual code to pair the DUT?

bzbarsky-apple, 2022-07-25:
> But if we set a long timeout, the chip-tool will repeat to establish the PASESession several times with the DUT

Only if it discovers the DUT multiple times (at different IPs, or on different interfaces or whatnot), no?  And if it does, failure on one of those does not mean the others will fail, though we could perhaps be smarter about guessing when they will.

> and when the chip-tool fails to establish the PASESession 20 times

How did we end up discovering the same thing 20 times?  Or is it that we're just getting to 20 failed PASE attempts before we have run the pairing command 20 times, because we discover it more than once (but not 20 times) each time?

Why are we ending up discovering the DUT multiple times at all?  What DUT is being used here and what do the advertisements look like?

> Delete the timeout and count the times of establishing the PASESession until it reaches 20 and then use the right manual code to pair the DUT

That would not be an SDK change, and should get filed in https://github.com/CHIP-Specifications/chip-test-plans/issues/new/choose

> Or stop the commissionable devices discovery once failing to establish the PASESession.

The SDK continues discovery, because discriminator collision can mean that there are multiple discovered devices and the PASE failure just means we contacted the wrong device...

wqx6, 2022-07-26:
@bzbarsky-apple I checked again on ESP32 after disabling IPv4 on it, the advertisements after open the commissioning window   is
```
$ avahi-browse -rt _matterc._udp
+ enp2s0 IPv6 50E8EFACE892F4ED                              _matterc._udp        local
= enp2s0 IPv6 50E8EFACE892F4ED                              _matterc._udp        local
   hostname = [7CDFA161C704.local]
   address = [fdde:ad00:beef:cafe:7edf:a1ff:fe61:c704]
   port = [5540]
   txt = ["PI=" "PH=36" "CM=2" "D=3840" "T=1" "SAI=300" "SII=5000" "VP=65521+32769"]
```
and the chip-tool will try to establish the PASESession with the ESP32 only once each time I send the command. But when I enable IPv4 on ESP32, the advertisements will on both IPv4 addr and IPv6 addr then the chip-tool will establish the PASESession with ESP32 multiple times.

Looks like we should disable IPv4 for our platform. 
>The SDK continues discovery, because discriminator collision can mean that there are multiple discovered devices and the PASE failure just means we contacted the wrong device...

BTW, could we add an option for the chip-tool, for example, `--discovery-once 1`, so that it will not discovery the device when it fails to establish the PASESession once.

bzbarsky-apple, 2022-07-26:
> then the chip-tool will establish the PASESession with ESP32 multiple times.

Sure, because it has no obvious way to tell that the different IPs are the same entity (well, apart from guessing at that based on the instance name) or knowing which ones will actually be able to reach the other side.

> BTW, could we add an option for the chip-tool

We probably could, yes.  We'd need to thread that down into the SetUpCodePairer, I suspect....

wqx6, 2022-08-03:
@bzbarsky-apple Is there any plan to solve this issue?

bzbarsky-apple, 2022-08-03:
I have no plans right now to work on it personally, but if someone wants to add the "discover only one thing, then stop" option, I would not object at all.

closed 2022-08-10 by wqx6