project-chip/connectedhomeip #21157: Add groups command fails

jadhavrohit924, 2022-07-25:
#### Problem
What is being fixed? Examples:

* Add groups command failing with
`AddGroupResponse:` {
[1658742473334] [28062:2229447] CHIP: [TOO] `status: 126`

[1658742473334] [28062:2229447] CHIP: [TOO] groupId: 1

[1658742473334] [28062:2229447] CHIP: [TOO] }

* Read group-table shows 0 entries 
` [1658745983622] [35924:2285458] CHIP: [TOO]   GroupTable: 0 entries`

 #### Proposed Solution
GroupAdd api fails here https://github.com/project-chip/connectedhomeip/blob/master/src/app/clusters/groups-server/groups-server.cpp#L113. Add suitable check.

bzbarsky-apple, 2022-07-25:
`126` is `UNSUPPORTED_ACCESS`.  @jadhavrohit924 Do you have a group key defined for the group id you are adding?  Because it sure sounds like you don't, and then per spec the `AddGroup` will fail with `UNSUPPORTED_ACCESS`.

Next time, please attach (not paste) complete logs starting with device commissioning to help with diagnosing issues like this; if I knew which exact commands you had run, I could tell you for sure whether your problem is missing group keys.

jadhavrohit924, 2022-07-25:
Sure I'll attach logs.
@bzbarsky-apple 
You are right I didn’t define group key. I directly used `AddGroup` command after commisssioning.

bzbarsky-apple, 2022-07-25:
In that case the observed behavior is correct.

closed 2022-07-25 by bzbarsky-apple