project-chip/connectedhomeip #21160: Chip-tool wrongly handle discovery via BLE with manual pairing code

doublemis1, 2022-07-25:
#### Problem
The Manual Pairing code contains the information about 4 most significant bits of discriminator.
The current implementation of chip-tool commissioning via BLE wrongly handles the discovery of devices that have set the discriminator to e.g. E64 (3684). The commissioning failed at BLE discovery since the manual paring code contains the E (E00 = 3584).

Parsed device manual pairing code:
```
chip_tool/chip-tool payload parse-setup-payload 34069117130
[1658752774.437685][17773:17773] CHIP:SPL: Parsing decimalRepresentation: 34069117130
[1658752774.437713][17773:17773] CHIP:SPL: Version:       0
[1658752774.437715][17773:17773] CHIP:SPL: VendorID:      0
[1658752774.437717][17773:17773] CHIP:SPL: ProductID:     0
[1658752774.437719][17773:17773] CHIP:SPL: Custom flow:   0    (STANDARD)
[1658752774.437720][17773:17773] CHIP:SPL: Capabilities:  0x00 (NONE)
[1658752774.437722][17773:17773] CHIP:SPL: Discriminator: 3584
[1658752774.437724][17773:17773] CHIP:SPL: Passcode:      28073715
```
Parsed device QR code payload:
```
chip-tool payload parse-setup-payload MT:KAYA3E.N16WWU-5KM00
[1658752813.867883][17774:17774] CHIP:SPL: Parsing base38Representation: MT:KAYA3E.N16WWU-5KM00
[1658752813.867921][17774:17774] CHIP:SPL: Version:       0
[1658752813.867924][17774:17774] CHIP:SPL: VendorID:      4735
[1658752813.867926][17774:17774] CHIP:SPL: ProductID:     32781
[1658752813.867928][17774:17774] CHIP:SPL: Custom flow:   0    (STANDARD)
[1658752813.867930][17774:17774] CHIP:SPL: Capabilities:  0x02 (BLE)
[1658752813.867933][17774:17774] CHIP:SPL: Discriminator: 3654
[1658752813.867936][17774:17774] CHIP:SPL: Passcode:      28073715
```

 #### Proposed Solution
The chip-tool during commissioning by manual pairing code shall connect the device based only on 4 most significant bits rather than the whole discriminator.

bzbarsky-apple, 2022-07-25:
@doublemis1 Which chip-tool are you running?  Linux or Mac?

The Mac one tries to handle this, in `checkDiscriminator` in `src/platform/Darwin/BleConnectionDelegateImpl.mm`.  It's not ideal in that it can incorrectly decide there's a match when we have a 12-bit discriminator but the low 8 bits are all zero, if the high 4 bits happen to match the high 4 bits of the device discriminator, but it should not fail to detect matches.

The Linux version (`BLEManagerImpl::OnDeviceScanned` in src/platform/Linux/BLEManagerImpl.cpp) does in fact look buggy.

doublemis1, 2022-07-25:
Yes, I used the Linux version.

closed 2022-07-26 by bzbarsky-apple