project-chip/connectedhomeip #21171: RaspberryPi Python Commissioning Consistently Segfaults

caubut-charter, 2022-07-25:
#### Problem

**Expected Behavior**

Expect the commissioning flow to start.

**Actual Behavior**

Consistently segfaults every time.

```
** (process:20003): CRITICAL **: 06:32:07.338: bluez_object_skeleton_new: assertion 'g_variant_is_object_path (object_path)' failed
2022-07-24 06:32:07 raspberrypi root[20003] WARNING  sys:1: Warning:g_variant_new_string: assertion 'string != NULL' failed
2022-07-24 06:32:07 raspberrypi root[20003] WARNING  sys:1: Warning:g_object_set: assertion 'G_IS_OBJECT (object)' failed

(process:20003): GLib-GIO-CRITICAL **: 06:32:07.345: g_dbus_object_manager_server_export: assertion 'G_IS_DBUS_OBJECT_MANAGER_SERVER (manager)' failed
2022-07-24 06:32:07 raspberrypi root[20003] WARNING  sys:1: Warning:g_object_unref: assertion 'G_IS_OBJECT (object)' failed
[New Thread 0x7fe37fe1c0 (LWP 20014)]
[Thread 0x7fe37fe1c0 (LWP 20014) exited]
2022-07-24 06:32:07 raspberrypi chip.BLE[20003] INFO BLE removing known devices.
2022-07-24 06:32:07 raspberrypi chip.BLE[20003] INFO BLE initiating scan.
2022-07-24 06:32:07 raspberrypi chip.DL[20003] INFO CHIPoBLE advertising config complete
2022-07-24 06:32:07 raspberrypi root[20003] WARNING  sys:1: Warning:g_variant_new_object_path: assertion 'g_variant_is_object_path (object_path)' failed

Thread 4 "python" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7ff2a011c0 (LWP 20008)]
0x0000007ff4227620 in ?? () from /lib/aarch64-linux-gnu/libglib-2.0.so.0
(gdb) bt
#0  0x0000007ff4227620 in  () at /lib/aarch64-linux-gnu/libglib-2.0.so.0
#1  0x0000007ff4223580 in g_variant_builder_add_value () at /lib/aarch64-linux-gnu/libglib-2.0.so.0
#2  0x0000007ff42254f8 in  () at /lib/aarch64-linux-gnu/libglib-2.0.so.0
#3  0x0000007ff4225bac in g_variant_new_va () at /lib/aarch64-linux-gnu/libglib-2.0.so.0
#4  0x0000007ff4225cec in g_variant_new () at /lib/aarch64-linux-gnu/libglib-2.0.so.0
#5  0x0000007ff4b66f68 in bluez_leadvertising_manager1_call_register_advertisement
    (proxy=0x7fe800f360, arg_advertisement=0x7fe802af60 "(null)/advertising", arg_options=0x7fe403ed90, cancellable=0x0, callback=0x7ff4b0a458 <chip::DeviceLayer::Internal::BluezAdvStartDone(GObject*, GAsyncResult*, gpointer)>, user_data=0x7fdc01c0b0) at gen/include/platform/Linux/dbus/bluez/DbusBluez.c:17249
#6  0x0000007ff4b0a8e4 in chip::DeviceLayer::Internal::BluezAdvStart(chip::DeviceLayer::Internal::BluezEndpoint*)
    (endpoint=0x7fdc01c0b0) at ../../src/platform/Linux/bluez/Helper.cpp:290
#7  0x0000007ff41e2ab4 in g_main_context_dispatch () at /lib/aarch64-linux-gnu/libglib-2.0.so.0
#8  0x0000007ff41e2e5c in  () at /lib/aarch64-linux-gnu/libglib-2.0.so.0
#9  0x0000007ff41e31b0 in g_main_loop_run () at /lib/aarch64-linux-gnu/libglib-2.0.so.0
#10 0x0000007ff4b03300 in chip::DeviceLayer::(anonymous namespace)::GDBus_Thread() ()
    at ../../src/platform/Linux/PlatformManagerImpl.cpp:68
#11 0x0000007ff4b051e0 in std::__invoke_impl<void, void (*)()>(std::__invoke_other, void (*&&)())
    (__f=@0x7fe4006c68: 0x7ff4b032e0 <chip::DeviceLayer::(anonymous namespace)::GDBus_Thread()>)
    at /usr/include/c++/10/bits/invoke.h:60
#12 0x0000007ff4b0516c in std::__invoke<void (*)()>(void (*&&)())
    (__fn=@0x7fe4006c68: 0x7ff4b032e0 <chip::DeviceLayer::(anonymous namespace)::GDBus_Thread()>)
    at /usr/include/c++/10/bits/invoke.h:95
#13 0x0000007ff4b05108 in std::thread::_Invoker<std::tuple<void (*)()> >::_M_invoke<0ul>(std::_Index_tuple<0ul>)
    (this=0x7fe4006c68) at /usr/include/c++/10/thread:264
#14 0x0000007ff4b050dc in std::thread::_Invoker<std::tuple<void (*)()> >::operator()() (this=0x7fe4006c68)
    at /usr/include/c++/10/thread:271
#15 0x0000007ff4b050bc in std::thread::_State_impl<std::thread::_Invoker<std::tuple<void (*)()> > >::_M_run()
    (this=0x7fe4006c60) at /usr/include/c++/10/thread:215
#16 0x0000007ff4080cac in  () at /lib/aarch64-linux-gnu/libstdc++.so.6
#17 0x0000007ff7fa3628 in start_thread (arg=0x7ff2a00ac0) at pthread_create.c:477
#18 0x0000007ff7dc401c in thread_start () at ../sysdeps/unix/sysv/linux/aarch64/clone.S:78
```

**Steps to Reproduce**

- https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/python_chip_controller_building.md#building-and-installing
- Edit `out/python_env/bin/chip-device-ctrl` to include a vendor: `self.fabricAdmin = FabricAdmin.FabricAdmin(0XFFF1)`
- `source out/python_env/bin/activate`
- `sudo out/python_env/bin/chip-device-ctrl`
- `chip-device-ctrl > ble-scan`
- `chip-device-ctrl > set-pairing-thread-credential 0e080000000000010000000300001335060004001fffe002084fe76e9a8b5edaf50708fde46f999f0698e20510d47f5027a414ffeebaefa92285cc84fa030f4f70656e5468726561642d653439630102e49c0410b92f8c7fbb4f9f3e08492ee3915fbd2f0c0402a0fff8`
- `chip-device-ctrl > connect -ble 3840 20202021 1234`

**System Configuration**
RaspberryPi 4b /w 64-bit RaspberryPi OS 

 #### Proposed Solution

The issue was introduced in this commit: https://github.com/project-chip/connectedhomeip/commit/a7a70ae4cabaece900ed9ea95d29f3e00b9d2dd3.

From the backtrace, it can been seen an invalid glib object path is being used of `"(null)/advertising"`:

```
#5  0x0000007ff4b66f68 in bluez_leadvertising_manager1_call_register_advertisement
    (proxy=0x7fe800f360, arg_advertisement=0x7fe802af60 "(null)/advertising", arg_options=0x7fe403ed90, cancellable=0x0, callback=0x7ff4b0a458 <chip::DeviceLayer::Internal::BluezAdvStartDone(GObject*, GAsyncResult*, gpointer)>, user_data=0x7fdc01c0b0) at gen/include/platform/Linux/dbus/bluez/DbusBluez.c:17249
```

It doesn't look like this line is getting called which initializes that value showing as null: https://github.com/project-chip/connectedhomeip/blob/2557f51c7ce40c4289f40fbecd001c0871b082a8/src/platform/Linux/bluez/Helper.cpp#L1309

I haven't been able to run down why that line is never being called as a result of the changes in the linked commit.

[pi-segfault.txt](https://github.com/project-chip/connectedhomeip/files/9183519/pi-segfault.txt)

bzbarsky-apple, 2022-07-25:
@mrjerryjohns can you please take a look?

mrjerryjohns, 2022-07-26:
@caubut-charter, the Python device-ctrl is in the process of being deprecated, and the REPL should be used instead.

The steps to commission a Thread device can be found [here](https://deepnote.com/viewer/github/project-chip/connectedhomeip/blob/master/docs/guides/repl/Matter%20-%20Basic%20Interactions.ipynb).

caubut-charter, 2022-07-26:
@mrjerryjohns Same issue using the REPL.

```
git clone https://github.com/project-chip/connectedhomeip.git
cd connectedhomeip
git submodule update --init
scripts/build_python.sh -m platform -i separate
source out/python_env/bin/activate
rm -rf /tmp/cihp*
hciconfig -a | tee pi-segfault2.txt
gdb --args python out/python_env/bin/chip-repl 2>&1 | tee pi-segfault2.txt
run
devCtrl.CommissionThread(3840, 20202021, 2, b'\x01\x03\xff')
bt
```

```
In [1]: devCtrl.CommissionThread(3840, 20202021, 2, b'\x01\x03\xff')
[New Thread 0x7fdffff1c0 (LWP 36874)]

** (process:36864): CRITICAL **: 21:15:14.695: bluez_object_skeleton_new: assertion 'g_variant_is_object_path (object_path)' failed

(process:36864): GLib-CRITICAL **: 21:15:14.698: g_variant_new_string: assertion 'string != NULL' failed

(process:36864): GLib-CRITICAL **: 21:15:14.698: g_variant_new_string: assertion 'string != NULL' failed

(process:36864): GLib-GObject-CRITICAL **: 21:15:14.699: g_object_set: assertion 'G_IS_OBJECT (object)' failed

(process:36864): GLib-GIO-CRITICAL **: 21:15:14.699: g_dbus_object_manager_server_export: assertion 'G_IS_DBUS_OBJECT_MANAGER_SERVER (manager)' failed

(process:36864): GLib-GObject-CRITICAL **: 21:15:14.699: g_object_unref: assertion 'G_IS_OBJECT (object)' failed

(process:36864): GLib-CRITICAL **: 21:15:14.718: g_variant_new_object_path: assertion 'g_variant_is_object_path (object_path)' failed

Thread 5 "python" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fee0f81c0 (LWP 36870)]
0x0000007fef846620 in ?? () from /lib/aarch64-linux-gnu/libglib-2.0.so.0
(gdb) bt
#0  0x0000007fef846620 in  () at /lib/aarch64-linux-gnu/libglib-2.0.so.0
#1  0x0000007fef842580 in g_variant_builder_add_value ()
    at /lib/aarch64-linux-gnu/libglib-2.0.so.0
#2  0x0000007fef8444f8 in  () at /lib/aarch64-linux-gnu/libglib-2.0.so.0
#3  0x0000007fef844bac in g_variant_new_va () at /lib/aarch64-linux-gnu/libglib-2.0.so.0
#4  0x0000007fef844cec in g_variant_new () at /lib/aarch64-linux-gnu/libglib-2.0.so.0
#5  0x0000007fefef368c in bluez_leadvertising_manager1_call_register_advertisement
    (proxy=0x7fe4015b70, arg_advertisement=0x7fe401d640 "(null)/advertising", arg_options=0x7fd802aef0, cancellable=0x0, callback=0x7fefe96b7c <chip::DeviceLayer::Internal::BluezAdvStartDone(GObject*, GAsyncResult*, gpointer)>, user_data=0x7fd801f020)
    at gen/include/platform/Linux/dbus/bluez/DbusBluez.c:17249
#6  0x0000007fefe97008 in chip::DeviceLayer::Internal::BluezAdvStart(chip::DeviceLayer::Internal::BluezEndpoint*) (endpoint=0x7fd801f020)
    at ../../src/platform/Linux/bluez/Helper.cpp:290
#7  0x0000007fef801ab4 in g_main_context_dispatch ()
    at /lib/aarch64-linux-gnu/libglib-2.0.so.0
#8  0x0000007fef801e5c in  () at /lib/aarch64-linux-gnu/libglib-2.0.so.0
#9  0x0000007fef8021b0 in g_main_loop_run () at /lib/aarch64-linux-gnu/libglib-2.0.so.0
#10 0x0000007fefe8fa2c in chip::DeviceLayer::(anonymous namespace)::GDBus_Thread() ()
    at ../../src/platform/Linux/PlatformManagerImpl.cpp:68
#11 0x0000007fefe91904 in std::__invoke_impl<void, void (*)()>(std::__invoke_other, void (*&&)())
    (__f=@0x7fe0006ba8: 0x7fefe8fa0c <chip::DeviceLayer::(anonymous namespace)::GDBus_Thread()>) at /usr/include/c++/10/bits/invoke.h:60
#12 0x0000007fefe91890 in std::__invoke<void (*)()>(void (*&&)())
    (__fn=@0x7fe0006ba8: 0x7fefe8fa0c <chip::DeviceLayer::(anonymous namespace)::GDBus_Thread()>) at /usr/include/c++/10/bits/invoke.h:95
#13 0x0000007fefe9182c in std::thread::_Invoker<std::tuple<void (*)()> >::_M_invoke<0ul>(std::_Index_tuple<0ul>) (this=0x7fe0006ba8) at /usr/include/c++/10/thread:264
#14 0x0000007fefe91800 in std::thread::_Invoker<std::tuple<void (*)()> >::operator()()
    (this=0x7fe0006ba8) at /usr/include/c++/10/thread:271
#15 0x0000007fefe917e0 in std::thread::_State_impl<std::thread::_Invoker<std::tuple<void (*)()> > >::_M_run() (this=0x7fe0006ba0) at /usr/include/c++/10/thread:215
#16 0x0000007fef69fcac in  () at /lib/aarch64-linux-gnu/libstdc++.so.6
#17 0x0000007ff7fa3628 in start_thread (arg=0x7fee0f7ac0) at pthread_create.c:477
#18 0x0000007ff7dc401c in thread_start () at ../sysdeps/unix/sysv/linux/aarch64/clone.S:78
quit)
```

[pi-segfault2.txt](https://github.com/project-chip/connectedhomeip/files/9193067/pi-segfault2.txt)

erjiaqing, 2022-07-27:
Looks like BLE is not running in central as expected?

erjiaqing, 2022-07-28:
@caubut-charter I have filed an PR to fix this issue and tested locally, please check if it works for you.

caubut-charter, 2022-07-28:
@erjiaqing I had to comment this out but otherwise it is working!

https://github.com/project-chip/connectedhomeip/blob/bfaa3904272105fa164d79f426cf267d8139df4b/src/controller/python/chip/native/__init__.py#L78-L79

It throws this exception if I don't.  I am not sure that condition would ever pass on the first call to `_GetLibraryHandle` with `shouldInit` set to `True`? 

```
(python_env) ➜  connectedhomeip git:(bfaa39042) python out/python_env/bin/chip-repl
Python 3.9.2 (default, Feb 28 2021, 17:03:44)
Type 'copyright', 'credits' or 'license' for more information
IPython 8.4.0 -- An enhanced Interactive Python. Type '?' for help.
---------------------------------------------------------------------------
Exception                                 Traceback (most recent call last)
File ~/test2/connectedhomeip/out/python_env/lib/python3.9/site-packages/chip/ChipReplStartup.py:14, in <module>
     12 import argparse
     13 import builtins
---> 14 import chip.FabricAdmin
     15 import chip.native
     16 from chip.utils import CommissioningBuildingBlocks

File ~/test2/connectedhomeip/out/python_env/lib/python3.9/site-packages/chip/FabricAdmin.py:36, in <module>
     32 from chip import ChipDeviceCtrl
     33 import copy
---> 36 class FabricAdmin:
     37     ''' Administers a given fabric instance. Each admin is associated with
     38         a specific RCAC and ICAC as well as Fabric ID and Index.
     39
   (...)
     64         boot for a given fabric and ensuring it automatically picks up the right ICAC/RCAC details as well.
     65     '''
     67     _handle = chip.native.GetLibraryHandle()

File ~/test2/connectedhomeip/out/python_env/lib/python3.9/site-packages/chip/FabricAdmin.py:67, in FabricAdmin()
     36 class FabricAdmin:
     37     ''' Administers a given fabric instance. Each admin is associated with
     38         a specific RCAC and ICAC as well as Fabric ID and Index.
     39
   (...)
     64         boot for a given fabric and ensuring it automatically picks up the right ICAC/RCAC details as well.
     65     '''
---> 67     _handle = chip.native.GetLibraryHandle()
     68     _isActive = False
     69     activeFabricIndexList = set()

File ~/test2/connectedhomeip/out/python_env/lib/python3.9/site-packages/chip/native/__init__.py:99, in GetLibraryHand
le()
     98 def GetLibraryHandle():
---> 99     return _GetLibraryHandle(True)

File ~/test2/connectedhomeip/out/python_env/lib/python3.9/site-packages/chip/native/__init__.py:79, in _GetLibraryHan
dle(shouldInit)
     77 if _nativeLibraryHandle is None:
     78     if shouldInit:
---> 79         raise Exception("Common stack has not been initialized!")
     80     _nativeLibraryHandle = ctypes.CDLL(FindNativeLibraryPath())
     81     setter = NativeLibraryHandleMethodArguments(_nativeLibraryHandle)

Exception: Common stack has not been initialized!
```

Once I commented that out, it commissioned just fine.

```
In [1]: devCtrl.CommissionThread(3840, 20202021, 2, bytes.fromhex("0e080000000000000000000300000b35060004001fffe00208
   ...: de7590f4b396e29d0708fddead00beef00000510296c8bcaf0a79a190a1e349a647e4b22030f537065637472756d2d356439393366010
   ...: 267490410cab1bb2db0906cb588e46f0629934f0f0c0402a0f7f8"))
2022-07-28 06:31:23 raspberrypi chip.DL[36794] ERROR Long dispatch time: 432 ms, for event type 2
Established secure session with Device
Commissioning complete
Out[1]: True
```

closed 2022-07-28 by woody-apple

erjiaqing, 2022-07-29:
> @erjiaqing I had to comment this out but otherwise it is working! [...]

Note: I also found this issue just after 21324, the fix should be included in the ToT already.

PR #21324 (2022-07-28): [Python] Fix BLE init sequence in Python REPL
#### Problem

Fixes #21171 

ConfigureBle should be called before  InitChipStack

#### Change overview

- Add chip.stack.Init()
- Move ConfigureBle to from `ChipStackInit` `pychip_CommonStackInit`

#### Testing
Manually tested, no crash