project-chip/connectedhomeip #21182: Fail to run bootstrap.sh on arm macos with failure in openssh

Ilidur, 2022-07-25:
#### Problem
<what's wrong or missing, please include any applicable:
 * expected behavior
 * actual behavior
 * steps to reproduce
 * system configuration
 * screenshots, images, logs, etc.
 >
When preparing to bootstrap building for the first time the ./scripts/bootstrap.sh fails on evaluating python environment. I've tracked down the issues to the following line which (includes an unguarded linux include pkg-config.)[https://github.com/project-chip/connectedhomeip/blob/251d49fcc42ead8694cae48b14fdbdc9d049363f/src/crypto/BUILD.gn#L51]

After putting some guards around it ` if (host_os == "linux")` the public_configs (here)[https://github.com/project-chip/connectedhomeip/blob/251d49fcc42ead8694cae48b14fdbdc9d049363f/src/crypto/BUILD.gn#L99] and (here)[https://github.com/project-chip/connectedhomeip/blob/1ff217418379280352c3f9047fef0ad88f68636c/src/tools/chip-cert/BUILD.gn#L39].

Commenting out those lines did not affect the completion of the build, however I do have openSSL installed on my mac via homebrew for unrelated reasons, so it may fail in other circumstances.

Note: I built all-clusters-app on an esp32C3 successfully.

 #### Proposed Solution
<suggested fix, suggested enhancement>
Handle dependencies for mac platform (at this line )[https://github.com/project-chip/connectedhomeip/blob/251d49fcc42ead8694cae48b14fdbdc9d049363f/src/crypto/BUILD.gn#L51]

Happy to help but I couldn't find a clean way to do it in the rest of the documentation.

bzbarsky-apple, 2022-07-25:
@Ilidur The build instructions say you have to install pkg-config on Mac to build.  Did you do that?

Ilidur, 2022-07-26:
Thanks for the support @bzbarsky-apple. I went straight to the example app build folder for my chip so I didn't notice there were prerequisite build instructions, as they also call `source ./scripts/bootstrap.sh`. Is it worth putting something in [there](https://github.com/project-chip/connectedhomeip/tree/master/examples/all-clusters-app/esp32)?

closed 2022-07-26 by Ilidur

bzbarsky-apple, 2022-07-26:
Sounds like it might be a good idea, yes.  @dhrishi, who owns the esp32 build instructions?

dhrishi, 2022-07-29:
Thanks for reporting the isue @Ilidur . @shubhamdp Please address this in your upcoming documentation PR, a pointer to the appropriate section in BUILDING.md for Mac as well as Linux