project-chip/connectedhomeip #21221: Build fails

sowmyassp, 2022-07-26:
**Issue :** 

Build is failing when I pass this command 

> ./scripts/run_in_build_env.sh "./scripts/build/build_examples.py --target linux-arm64-all-clusters-no-ble-asan-libfuzzer-clang build"

[image]

**Commit used**: 29873290196796fbf23e28083163f8565bb479ba
**Chip-tool** - RPI-4, 8GB RAM
**DUT** - RPI - RPI-4, 8GB RAM
**Platform** : linux platform
**Reference** :
https://github.com/project-chip/connectedhomeip/tree/master/examples/all-clusters-app/linux#readme

bzbarsky-apple, 2022-07-26:
@sowmyassp Please attach (not paste) the full log from the compilation.  The screenshot there cuts off what the actual error is; it's higher up somewhere.

sowmyassp, 2022-07-26:
@bzbarsky-apple 
[Log.txt](https://github.com/project-chip/connectedhomeip/files/9190946/Log.txt)

bzbarsky-apple, 2022-07-26:
Relevant errors:
```
2022-07-26 15:04:45 INFO    ld.lld: error: duplicate symbol: main
2022-07-26 15:04:45 INFO    >>> defined at FuzzerMain.cpp:19 (compiler-rt/lib/fuzzer/FuzzerMain.cpp:19)
2022-07-26 15:04:45 INFO    >>>            fuzzer.o:(main) in archive /home/ubuntu/jul26_dut/connectedhomeip/.environment/cipd/packages/pigweed/lib/clang/14.0.0/lib/aarch64-unknown-linux-gnu/libclang_rt.fuzzer.a
2022-07-26 15:04:45 INFO    >>> defined at tool.cpp:136 (../../examples/all-clusters-app/linux/third_party/connectedhomeip/src/lib/address_resolve/tool.cpp:136)
2022-07-26 15:04:45 INFO    >>>            obj/third_party/connectedhomeip/src/lib/address_resolve/address-resolve-tool.tool.cpp.o:(.text.main+0x0)
2022-07-26 15:04:45 INFO    clang-14: error: linker command failed with exit code 1 (use -v to see invocation)
```
Why is this building `address-resolve-tool` at all?

@sowmyassp What is the output from:
```
    ./scripts/run_in_build_env.sh "./scripts/build/build_examples.py --dry-run --target linux-arm64-all-clusters-no-ble-asan-libfuzzer-clang build"
```
for you?

sowmyassp, 2022-07-26:
[image]

bzbarsky-apple, 2022-07-26:
@sowmyassp Please attach complete logs; you cut off the actual error again.  But also: did you actually pass `--dry-run` there?  It sure does not look like you did.

sowmyassp, 2022-07-26:
@bzbarsky-apple you mean to say I need to send 
./scripts/run_in_build_env.sh "./scripts/build/build_examples.py --target linux-arm64-all-clusters-no-ble-asan-libfuzzer-clang build" --dry-run

PFA Log:
[Log.txt](https://github.com/project-chip/connectedhomeip/files/9191706/Log.txt)

bzbarsky-apple, 2022-07-26:
No, you need to do the thing I said in https://github.com/project-chip/connectedhomeip/issues/21221#issuecomment-1195662096.  Please just copy and paste that exact command.

sowmyassp, 2022-07-27:
@bzbarsky-apple Here is the log for above mentioned command
[log_1.txt](https://github.com/project-chip/connectedhomeip/files/9195227/log_1.txt)

cjandhyala, 2022-07-27:
@bzbarsky-apple I was trying to build the same on my linux desktop I got "ld.lld: error: duplicate symbol: main" while building address-resolve-tool. Attached log from my machine

[log.txt](https://github.com/project-chip/connectedhomeip/files/9195329/log.txt)

yufengwangca, 2022-07-27:
It looks libfuzzer also defined main function at lib/fuzzer/FuzzerMain.cpp:19 which is conflict with the main function defined in our example app

```
2022-07-27 10:46:35 INFO    ld.lld: error: duplicate symbol: main
2022-07-27 10:46:35 INFO    >>> defined at FuzzerMain.cpp:19 (compiler-rt/lib/fuzzer/FuzzerMain.cpp:19
```

woody-apple, 2022-07-27:
Cert Blocker Review: We should update instructions here for the CSG. @bzbarsky-apple @mspang can you take a peek?

bzbarsky-apple, 2022-07-27:
@yufengwangca Yes, libfuzzer defines a different `main`.  Building `all-clusters-app` excludes `main` when building with libfuzzer.  The problem is the `address-resolve-tool` that is getting built alongside `all-clusters-app`  It should not be getting built.  @mspang @andy31415 Why is that getting built?

closed 2022-07-28 by woody-apple