project-chip/connectedhomeip #21227: Provisioning not working on M1 MacBook

rosahay-silabs, 2022-07-26:
#### Problem
Commission not working on the macbook M1 Pro.

 * expected behavior
     User should be able to commission DUT by using the chip-tool on the M1 macbook.
 * actual behavior
     User unable to commission the DUT.
 * steps to reproduce
     - Compile chip-tool on device.
     - Put device in the BLE advertisement mode.
     - Start commissioning process on macbook.
 * system configuration:
     - MacBook Pro M1(Apple silicon)/macOS Monterey(12.2.1)
     - RS9116+MG24, device under test(DUT)

[macbook.chip-tool.run.log](https://github.com/project-chip/connectedhomeip/files/9189721/macbook.chip-tool.run.log)

bzbarsky-apple, 2022-07-26:
@woody-apple Is 12.2.1 one of the MacOS versions with issues around BLE commissioning with chip-tool?

woody-apple, 2022-07-27:
Yes please, @rosahay-silabs can you try the latest macOS release? There were some known BLE issues on previous releases

woody-apple, 2022-07-27:
Please re-open if this happens on the newest releases

closed 2022-07-27 by woody-apple