project-chip/connectedhomeip #21236: [TC-OCC-3.2] all-clusters-app has no functionality to change occupancy status

s-mcclain, 2022-07-26:
#### Problem
Occupancy Sensing cluster test case TC-OCC-3.2 still requires the Test Harness (all-cluster-app, etc.) to change the occupancy status in some way. This is not currently supported in the current applications which makes it impossible to execute this test case where the DUT is the client.

 #### Proposed Solution
Enable some sort of argument/mechanism for the all-cluster-app to be able to change the occupancy status on the application.

woody-apple, 2022-07-27:
Cert Blocker Review: Need to find an owner for this. Lower priority given this is optional.

woody-apple, 2022-07-29:
Cert Blocker Review: Marking this as 'needs provisional' unless we have a resolution to this issue today.

closed 2023-02-12 by stale[bot]